chipsec/chipsec #1572: Infinite Loop when detecting UEFI volumes with bad FV Size

platomav, 2022-08-31:
Chipsec seems to hang when trying to detect potential UEFI volumes at some files. The issue occurs because of the following line:

https://github.com/chipsec/chipsec/blob/fd1a98688978fd4b8ca77b512a72eae49c3beffa/chipsec/hal/spi_uefi.py#L294

The parent function attempts to detect all UEFI volumes (tag _FVH) in a “while” loop without making sure that their reported Size field is valid (i.e. large enough). If the FV Size is smaller than the length of FV Header (total 0x38, 0x28 before _FVH, 0x2C including _FVH), the same FV will be detected again and again in a never ending loop.

A small sanity check can be added to make sure that the next FV Offset is actually after the current FV Header. For example:

```python
fv = NextFwVolume(data, fv.Offset + max(fv.Size, 0x2C))
```

The attached sample (EF0.bin) can be used to reproduce the infinite loop issue (after unzipping it):

[EF0.zip](https://github.com/chipsec/chipsec/files/9460696/EF0.zip)

The same oversight might exist at other similar functions.

platomav, 2022-08-31:
Issue #1173 seems to be the same as this one. Both samples show the same type of corruption in which the FV Size is stated as 0x8 which is smaller than the Header size (0x2C) so the same _FVH is detected in an infinite loop. Despite the samples being corrupted, the infinite loop should not occur.

platomav, 2022-12-02:
Thank you for the fix! Indeed, it's better to use the header size definition instead of a hard-coded value.

I can verify that the specific attached sample does work now when trying to retrieve its binaries.

However, the same function/logic is being used elsewhere with the potential to cause infinite loops in other cases as well:

[image]

dgutson, 2022-12-02:
Maybe the fix should be encapsulated inside the function, thus not requiring the caller to take care of this, and therefore having the fix in one place for all? @npmitche

npmitche, 2022-12-05:
I made the change in #1607 and moved the `max` call to the `NextFwVolume` function. If possible, please review and/or test and let me know if you have any other suggestions. Appreciate the help!

dgutson, 2022-12-05:
@platomav any chance to re-test it pls?

platomav, 2022-12-06:
Ok, I re-tested and the infinite loop is still fixed. The logic of the max call at NextFwVolume looks ok. In theory, a false positive "last_fv_size" of 0x0 (instead of 0x8 at the test sample), could cause the problem (at _FVH one after another with 0x0 size), but from what I can tell, the two checks within the while loop (break, continue) will avoid that. So, it seems to work now.

However, the latest commit (https://github.com/npmitche/chipsec/commit/139b5ee347934ee1bb0798acd0a423b92f1dcb03) crashes with a circular "chipsec.logger" module import.

``` python
Traceback (most recent call last):
  File "/app/my_app/my_app.py", line 11, in <module>
    from abstract_my_app import my_app
  File "/app/my_app/abstract_my_app.py", line 11, in <module>
    from chipsec.modules.tools.uefi.my_custom_module import ModuleResult, my_custom_module
  File "/app/my_app/chipsec/modules/tools/uefi/my_custom_module.py", line 22, in <module>
    from chipsec.module_common import BaseModule, ModuleResult, MTAG_BIOS
  File "/app/my_app/chipsec/module_common.py", line 26, in <module>
    import chipsec.logger
  File "/app/my_app/chipsec/logger.py", line 34, in <module>
    from chipsec.defines import get_version, get_message, os_version
  File "/app/my_app/chipsec/defines.py", line 27, in <module>
    import chipsec.file
  File "/app/my_app/chipsec/file.py", line 42, in <module>
    from chipsec.logger import logger
ImportError: cannot import name 'logger' from partially initialized module 'chipsec.logger' (most likely due to a circular import) (/app/my_app/chipsec/logger.py)
```

I can verify this is not the case with the latest official release (https://github.com/chipsec/chipsec/commit/d8c2a606bf440c32196c6289a7a458f3ae3107cc).

In general, the logger in Chipsec appears to misbehave because it "hijacks" (?) the logging module in a way which causes external (same interpreter instance) logs to not render properly. Here is pseudocode for that behavior:

```
import logging

my_logger = logging.getLogger("my_app")

my_logger.warning('This is a %s', 'test_1')

from chipsec.modules.tools.uefi import <something>

my_logger.warning('This is a %s', 'test_2')

> This is a test_1
> This is a %s
```

I'm not sure if the logging "hijack" issue occurred now or some time before.

npmitche, 2022-12-07:
It was caused by a recent change (#1604). Oddly enough, it doesn't show up when we run chipsec_main, but it is present and we'll fix it before the next release. Appreciate the report!

closed 2022-12-07 by npmitche

platomav, 2022-12-08:
> Yes, please open a separate ticket for that.

Done at #1617 and #1618, thank you for the fixes so far! 👍