chipsec/chipsec #1573: Failure to find all EFI binaries at Lenovo FL1 and/or FL2 update images

platomav, 2022-08-31:
Chipsec is not able to find all EFI images within some Lenovo "FL1" or "FL2" images when scanning for them via:

```bash
chipsec_main.py --skip_config --no_driver -n -m tools.uefi.scan_image -a generate $0AHJ000_FL2.json '/path/to/$0AHJ000.FL2'
```

The attached sample ($0AHJ000.FL2) can be used to reproduce the issue (after unzipping it):

[$0AHJ000.zip](https://github.com/chipsec/chipsec/files/9460795/0AHJ000.zip)

Opening the image in UEFITool, it consists of GUIDs “6C60EE00-C316-4C95-A684-CDC7E7033311” and “C8AB0F4E-26FE-40F1-9579-EA8D30D503A4” but Chipsec only detects “C8AB0F4E-26FE-40F1-9579-EA8D30D503A4”, which contains a Lenovo “flasher” image only, thus 1 result. However, “6C60EE00-C316-4C95-A684-CDC7E7033311” includes another 336 images which are not detected.

[image]

The expected result is for Chipsec to detect 337 images from the given sample.

npmitche, 2022-11-30:
Sorry for the long delay, but can you re-test using this PR: https://github.com/chipsec/chipsec/pull/1607

platomav, 2022-12-02:
Thank you for the fix! I can verify that the attached sample now produces the expected amount of binaries, meaning 337 instead of 1. Feel free to close this when the PR is ready.

npmitche, 2022-12-07:
Merged in #1607

closed 2022-12-07 by npmitche