chipsec/chipsec #1582: Cannot build manual on Linux Mint 20.3 (Ubuntu 20.04 LTS base)

hamishmb, 2022-09-27:
Hi there,

I've found I can't build the chipsec manual on Linux Mint 20.3, based on Ubuntu 20.04 LTS. This is with the distro-provided Python 3.8, and rst2pdf installed via pip.

I have attached the output from the docs/create_manual.sh command below.

[manual_log.log](https://github.com/chipsec/chipsec/files/9656422/manual_log.log)

If you need any more information, let me know.

Hamish

npmitche, 2022-09-27:
I don't see an error in the log, Can you try to delete (or move) the file "chipsec_manual.pdf" located in the root chipsec directory and re-run? Does the file show up there again?

hamishmb, 2022-09-27:
My bad, I forgot to redirect stderr to stdout, meaning that all the errors were missing from the log. Here's a new log with the errors in.

[manual_log.log](https://github.com/chipsec/chipsec/files/9657462/manual_log.log)

I can confirm that the file doesn't reappear for me after deleting it and running the script.

npmitche, 2022-10-12:
Can you try updating Sphinx? I see you're running v1.8.5. I am able to build it on v4.4.0.

hamishmb, 2022-10-25:
I can confirm that it works for me with the latest Sphinx from PyPI (5.3.0).

I do get this warning, though:

```
WARNING: Invalid configuration value found: 'language = None'. Update your configuration to a valid language code. Falling back to 'en' (English).
```

So I guess the problem here is more that there's no declared minimum version of Sphinx required to build the manual. 1.8.5 is only from 2019 (version from the Ubuntu 20.04 repos), so probably worth trying to figure out what the minimum requirement is somehow.

npmitche, 2022-11-30:
@Sae86 Can you investigate and document the minimum sphinx version needed for building our manual?

PR #2129 (2024-04-15): Add documentation for Sphinx including min and max versions
Fixes #1582
closed 2024-04-15 by dscott90