chipsec/chipsec #1597: Failed to Run vs Faild

seprj, 2022-11-01:
Hello
I have a Lenovo laptop, which I applied  chipset on it, there are 14 failed to Run, what is the difference between the faild and failed to Run?
and in faild to run I see "ERROR" I dont know whats the meaning of?
beside it I see 13 Modules with exceptions also I dont know whats the  meaning of this tooooo
thanks to answer to me

dscott90, 2022-11-07:
Hi,

A guide for interpreting results can be located on page 11 of the chipsec manual.
If you'd like to attach a copy of your output when running chipsec tool, I can help clarify some of the things you are seeing.

seprj, 2022-11-09:
Hi
Thanks for your respond and advise
my output when running chipsec is:
[CHIPSEC] ***************************  SUMMARY  ***************************
[CHIPSEC] Time elapsed            1.000
[CHIPSEC] Modules total           28
[CHIPSEC] Modules failed to run   14:
ERROR: chipsec.modules.common.bios_kbrd_buffer
ERROR: chipsec.modules.common.bios_smi
ERROR: chipsec.modules.common.bios_wp
ERROR: chipsec.modules.common.remap
ERROR: chipsec.modules.common.rtclock
ERROR: chipsec.modules.common.secureboot.variables
ERROR: chipsec.modules.common.smrr
ERROR: chipsec.modules.common.spi_access
ERROR: chipsec.modules.common.spi_desc
ERROR: chipsec.modules.common.spi_fdopss
ERROR: chipsec.modules.common.spi_lock
ERROR: chipsec.modules.common.uefi.access_uefispec
ERROR: chipsec.modules.common.uefi.s3bootscript
ERROR: chipsec.modules.common.wsmt
[CHIPSEC] Modules passed          9:
[+] PASSED: chipsec.modules.common.bios_ts
[+] PASSED: chipsec.modules.common.cpu.ia_untrusted
[+] PASSED: chipsec.modules.common.debugenabled
[+] PASSED: chipsec.modules.common.ia32cfg
[+] PASSED: chipsec.modules.common.me_mfg_mode
[+] PASSED: chipsec.modules.common.memconfig
[+] PASSED: chipsec.modules.common.smm_code_chk
[+] PASSED: chipsec.modules.common.smm_dma
[+] PASSED: chipsec.modules.common.spd_wd
[CHIPSEC] Modules information     1:
[#] INFORMATION: chipsec.modules.common.cpu.cpu_info
[CHIPSEC] Modules failed          0:
[CHIPSEC] Modules with warnings   1:
[!] WARNING: chipsec.modules.common.cpu.spectre_v2
[CHIPSEC] Modules not implemented 0:
[CHIPSEC] Modules not applicable  3:
[*] NOT APPLICABLE: chipsec.modules.common.memlock
[*] NOT APPLICABLE: chipsec.modules.common.sgx_check
[*] NOT APPLICABLE: chipsec.modules.common.smm
[CHIPSEC] Modules with exceptions 13:
ERROR: chipsec.modules.common.bios_kbrd_buffer
ERROR: chipsec.modules.common.bios_smi
ERROR: chipsec.modules.common.bios_wp
ERROR: chipsec.modules.common.remap
ERROR: chipsec.modules.common.rtclock
ERROR: chipsec.modules.common.secureboot.variables
ERROR: chipsec.modules.common.smrr
ERROR: chipsec.modules.common.spi_access
ERROR: chipsec.modules.common.spi_desc
ERROR: chipsec.modules.common.spi_fdopss
ERROR: chipsec.modules.common.spi_lock
ERROR: chipsec.modules.common.uefi.access_uefispec
ERROR: chipsec.modules.common.wsmt
[CHIPSEC] *****************************************************************

dscott90, 2022-11-10:
Hi @seprj,

Can you run the tool again using the -v argument and share the entire log file? (Chipsec can write directly to text files using the -l argument)
The -v argument adds verbosity to the logging so more data about the module runs is available.

seprj, 2022-11-13:
Hi @npmitche , thanks for your respond 
I run this tool using -v argument  then I have some Error , one of them is 

[*] running module: chipsec.modules.common.memlock
'MSR_LT_LOCK_MEMORY.LT_LOCK' not defined for platform.  Skipping module.
Skipping module chipsec.modules.common.memlock since it is not supported in this platform

[*] running module: chipsec.modules.common.remap
[x][ =======================================================================
[x][ Module: Memory Remapping Configuration
[x][ =======================================================================
ERROR: Exception occurred during chipsec.modules.common.remap.run(): 'argument must be sequence of length 2, not 3'

[*] running module: chipsec.modules.common.rtclock
[x][ =======================================================================
[x][ Module: Protected RTC memory locations
[x][ =======================================================================
ERROR: Exception occurred during chipsec.modules.common.rtclock.run(): 'argument must be sequence of length 2, not 3'

[*] running module: chipsec.modules.common.secureboot.variables
[x][ =======================================================================
[x][ Module: Attributes of Secure Boot EFI Variables
[x][ =======================================================================
ERROR: Exception occurred during chipsec.modules.common.secureboot.variables.run(): 'must be unicode, not str'

seprj, 2022-11-13:
@npmitche  
 this is My new summery:

[CHIPSEC] ***************************  SUMMARY  ***************************
[CHIPSEC] Time elapsed            1.000
[CHIPSEC] Modules total           28
[CHIPSEC] Modules failed to run   14:
ERROR: chipsec.modules.common.bios_kbrd_buffer
ERROR: chipsec.modules.common.bios_smi
ERROR: chipsec.modules.common.bios_wp
ERROR: chipsec.modules.common.remap
ERROR: chipsec.modules.common.rtclock
ERROR: chipsec.modules.common.secureboot.variables
ERROR: chipsec.modules.common.smrr
ERROR: chipsec.modules.common.spi_access
ERROR: chipsec.modules.common.spi_desc
ERROR: chipsec.modules.common.spi_fdopss
ERROR: chipsec.modules.common.spi_lock
ERROR: chipsec.modules.common.uefi.access_uefispec
ERROR: chipsec.modules.common.uefi.s3bootscript
ERROR: chipsec.modules.common.wsmt
[CHIPSEC] Modules passed          9:
[+] PASSED: chipsec.modules.common.bios_ts
[+] PASSED: chipsec.modules.common.cpu.ia_untrusted
[+] PASSED: chipsec.modules.common.debugenabled
[+] PASSED: chipsec.modules.common.ia32cfg
[+] PASSED: chipsec.modules.common.me_mfg_mode
[+] PASSED: chipsec.modules.common.memconfig
[+] PASSED: chipsec.modules.common.smm_code_chk
[+] PASSED: chipsec.modules.common.smm_dma
[+] PASSED: chipsec.modules.common.spd_wd
[CHIPSEC] Modules information     1:
[#] INFORMATION: chipsec.modules.common.cpu.cpu_info
[CHIPSEC] Modules failed          0:
[CHIPSEC] Modules with warnings   1:
[!] WARNING: chipsec.modules.common.cpu.spectre_v2
[CHIPSEC] Modules not implemented 0:
[CHIPSEC] Modules not applicable  3:
[*] NOT APPLICABLE: chipsec.modules.common.memlock
[*] NOT APPLICABLE: chipsec.modules.common.sgx_check
[*] NOT APPLICABLE: chipsec.modules.common.smm
[CHIPSEC] Modules with exceptions 14:
ERROR: chipsec.modules.common.bios_kbrd_buffer
ERROR: chipsec.modules.common.bios_smi
ERROR: chipsec.modules.common.bios_wp
ERROR: chipsec.modules.common.remap
ERROR: chipsec.modules.common.rtclock
ERROR: chipsec.modules.common.secureboot.variables
ERROR: chipsec.modules.common.smrr
ERROR: chipsec.modules.common.spi_access
ERROR: chipsec.modules.common.spi_desc
ERROR: chipsec.modules.common.spi_fdopss
ERROR: chipsec.modules.common.spi_lock
ERROR: chipsec.modules.common.uefi.access_uefispec
ERROR: chipsec.modules.common.uefi.s3bootscript
ERROR: chipsec.modules.common.wsmt
[CHIPSEC] *****************************************************************

seprj, 2022-11-13:
@npmitche  this is my new summery

[CHIPSEC] ***************************  SUMMARY  ***************************
[CHIPSEC] Time elapsed            1.000
[CHIPSEC] Modules total           28
[CHIPSEC] Modules failed to run   14:
ERROR: chipsec.modules.common.bios_kbrd_buffer
ERROR: chipsec.modules.common.bios_smi
ERROR: chipsec.modules.common.bios_wp
ERROR: chipsec.modules.common.remap
ERROR: chipsec.modules.common.rtclock
ERROR: chipsec.modules.common.secureboot.variables
ERROR: chipsec.modules.common.smrr
ERROR: chipsec.modules.common.spi_access
ERROR: chipsec.modules.common.spi_desc
ERROR: chipsec.modules.common.spi_fdopss
ERROR: chipsec.modules.common.spi_lock
ERROR: chipsec.modules.common.uefi.access_uefispec
ERROR: chipsec.modules.common.uefi.s3bootscript
ERROR: chipsec.modules.common.wsmt
[CHIPSEC] Modules passed          9:
[+] PASSED: chipsec.modules.common.bios_ts
[+] PASSED: chipsec.modules.common.cpu.ia_untrusted
[+] PASSED: chipsec.modules.common.debugenabled
[+] PASSED: chipsec.modules.common.ia32cfg
[+] PASSED: chipsec.modules.common.me_mfg_mode
[+] PASSED: chipsec.modules.common.memconfig
[+] PASSED: chipsec.modules.common.smm_code_chk
[+] PASSED: chipsec.modules.common.smm_dma
[+] PASSED: chipsec.modules.common.spd_wd
[CHIPSEC] Modules information     1:
[#] INFORMATION: chipsec.modules.common.cpu.cpu_info
[CHIPSEC] Modules failed          0:
[CHIPSEC] Modules with warnings   1:
[!] WARNING: chipsec.modules.common.cpu.spectre_v2
[CHIPSEC] Modules not implemented 0:
[CHIPSEC] Modules not applicable  3:
[*] NOT APPLICABLE: chipsec.modules.common.memlock
[*] NOT APPLICABLE: chipsec.modules.common.sgx_check
[*] NOT APPLICABLE: chipsec.modules.common.smm
[CHIPSEC] Modules with exceptions 14:
ERROR: chipsec.modules.common.bios_kbrd_buffer
ERROR: chipsec.modules.common.bios_smi
ERROR: chipsec.modules.common.bios_wp
ERROR: chipsec.modules.common.remap
ERROR: chipsec.modules.common.rtclock
ERROR: chipsec.modules.common.secureboot.variables
ERROR: chipsec.modules.common.smrr
ERROR: chipsec.modules.common.spi_access
ERROR: chipsec.modules.common.spi_desc
ERROR: chipsec.modules.common.spi_fdopss
ERROR: chipsec.modules.common.spi_lock
ERROR: chipsec.modules.common.uefi.access_uefispec
ERROR: chipsec.modules.common.uefi.s3bootscript
ERROR: chipsec.modules.common.wsmt
[CHIPSEC] *****************************************************************

dscott90, 2022-11-15:
Hi @seprj,

Attaching the entire log file rather than snippets of output will help me better understand your issues. I'd like to see the platform and argument data.

Thanks

seprj, 2022-11-16:
Hi @dscott90 , Thanks for your answer
This is entire log file

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version 1.8.0
[CHIPSEC] Arguments: -v

[CHIPSEC] API mode: using CHIPSEC kernel module API
[CHIPSEC] OS      : uefi   
[CHIPSEC] Python  : 2.7.2 (64-bit)
[CHIPSEC] Helper  : EfiHelper (None)
[CHIPSEC] Platform: TGL UP3 4 Cores
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 9A14
[CHIPSEC]      RID: 01
[CHIPSEC] PCH     : 5xx PCH Premium UP3
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: A082
[CHIPSEC]      RID: 20
 
[*] loading common modules from ".\chipsec\modules\common" ..
[*] No platform specific modules to load
[*] loading modules from ".\chipsec\modules" ..
[+] loaded chipsec.modules.common.bios_kbrd_buffer
[+] loaded chipsec.modules.common.bios_smi
[+] loaded chipsec.modules.common.bios_ts
[+] loaded chipsec.modules.common.bios_wp
[+] loaded chipsec.modules.common.cpu.cpu_info
[+] loaded chipsec.modules.common.cpu.ia_untrusted
[+] loaded chipsec.modules.common.cpu.spectre_v2
[+] loaded chipsec.modules.common.debugenabled
[+] loaded chipsec.modules.common.ia32cfg
[+] loaded chipsec.modules.common.me_mfg_mode
[+] loaded chipsec.modules.common.memconfig
[+] loaded chipsec.modules.common.memlock
[+] loaded chipsec.modules.common.remap
[+] loaded chipsec.modules.common.rtclock
[+] loaded chipsec.modules.common.secureboot.variables
[+] loaded chipsec.modules.common.sgx_check
[+] loaded chipsec.modules.common.smm
[+] loaded chipsec.modules.common.smm_code_chk
[+] loaded chipsec.modules.common.smm_dma
[+] loaded chipsec.modules.common.smrr
[+] loaded chipsec.modules.common.spd_wd
[+] loaded chipsec.modules.common.spi_access
[+] loaded chipsec.modules.common.spi_desc
[+] loaded chipsec.modules.common.spi_fdopss
[+] loaded chipsec.modules.common.spi_lock
[+] loaded chipsec.modules.common.uefi.access_uefispec
[+] loaded chipsec.modules.common.uefi.s3bootscript
[+] loaded chipsec.modules.common.wsmt
[*] running loaded modules ..

[*] running module: chipsec.modules.common.bios_kbrd_buffer
[x][ =======================================================================
[x][ Module: Pre-boot Passwords in the BIOS Keyboard Buffer
[x][ =======================================================================
ERROR: Exception occurred during chipsec.modules.common.bios_kbrd_buffer.run(): 'argument must be sequence of length 2, not 3'

[*] running module: chipsec.modules.common.bios_smi
[x][ =======================================================================
[x][ Module: SMI Events Configuration
[x][ =======================================================================
[*] BC = 0x100008AA << BIOS Control (b:d.f 00:31.5 + 0xDC)
    [00] BIOSWE           = 0 << BIOS Write Enable 
    [01] BLE              = 1 << BIOS Lock Enable 
[... 415 lines cut ...]
[#] INFORMATION: chipsec.modules.common.cpu.cpu_info
[CHIPSEC] Modules failed          0:
[CHIPSEC] Modules with warnings   1:
[!] WARNING: chipsec.modules.common.cpu.spectre_v2
[CHIPSEC] Modules not implemented 0:
[CHIPSEC] Modules not applicable  3:
[*] NOT APPLICABLE: chipsec.modules.common.memlock
[*] NOT APPLICABLE: chipsec.modules.common.sgx_check
[*] NOT APPLICABLE: chipsec.modules.common.smm
[CHIPSEC] Modules with exceptions 14:
ERROR: chipsec.modules.common.bios_kbrd_buffer
ERROR: chipsec.modules.common.bios_smi
ERROR: chipsec.modules.common.bios_wp
ERROR: chipsec.modules.common.remap
ERROR: chipsec.modules.common.rtclock
ERROR: chipsec.modules.common.secureboot.variables
ERROR: chipsec.modules.common.smrr
ERROR: chipsec.modules.common.spi_access
ERROR: chipsec.modules.common.spi_desc
ERROR: chipsec.modules.common.spi_fdopss
ERROR: chipsec.modules.common.spi_lock
ERROR: chipsec.modules.common.uefi.access_uefispec
ERROR: chipsec.modules.common.uefi.s3bootscript
ERROR: chipsec.modules.common.wsmt
[CHIPSEC] *****************************************************************

dscott90, 2022-11-17:
@seprj thanks for your reply.

I'm seeing that you are currently running chipsec using python 2.7. Python 2 has been deprecated and python 3 is now required to run chipsec successfully.

The log also shows that the chipsec version you're using is 1.8.0. We are currently on release 1.9.1 and you may have better results using that latest version.

Could you please update your python installation, try running chipsec release 1.9.1, and share your output with me again?

Thanks,
Dan

seprj, 2022-11-27:
Thanks for your help :)

closed 2022-11-28 by dscott90

seprj, 2023-01-09:
Hello
can you help me about Rtclock
In Chipsec, in the rtclock module, I know that the third bit of RC register, which is the result of LL and UL, if only one of them(LL/UL) is 1, causes Rtclock memory not to be locked, how can third bit be always set to 1 through Chipsec and always Rtclock keep locked
thank you for your guide