chipsec/chipsec #1617: Circular import crash at "chipsec.logger"

platomav, 2022-12-08:
The latest commit (7d97b92a27e3f52175a082e602c4ef75481bd6e2, https://github.com/chipsec/chipsec/pull/1607) crashes with a circular "chipsec.logger" module import.

``` python
Traceback (most recent call last):
  File "/app/my_app/my_app.py", line 11, in <module>
    from abstract_my_app import my_app
  File "/app/my_app/abstract_my_app.py", line 11, in <module>
    from chipsec.modules.tools.uefi.my_custom_module import ModuleResult, my_custom_module
  File "/app/my_app/chipsec/modules/tools/uefi/my_custom_module.py", line 22, in <module>
    from chipsec.module_common import BaseModule, ModuleResult, MTAG_BIOS
  File "/app/my_app/chipsec/module_common.py", line 26, in <module>
    import chipsec.logger
  File "/app/my_app/chipsec/logger.py", line 34, in <module>
    from chipsec.defines import get_version, get_message, os_version
  File "/app/my_app/chipsec/defines.py", line 27, in <module>
    import chipsec.file
  File "/app/my_app/chipsec/file.py", line 42, in <module>
    from chipsec.logger import logger
ImportError: cannot import name 'logger' from partially initialized module 'chipsec.logger' (most likely due to a circular import) (/app/my_app/chipsec/logger.py)
```

According to https://github.com/chipsec/chipsec/issues/1572#issuecomment-1341239737, it was caused by a recent PR (https://github.com/chipsec/chipsec/pull/1604).

Opening this as a separate issue to be tracked accordingly.

Sae86, 2022-12-08:
This should be fixed in PR #1616

platomav, 2022-12-09:
``` python
[...]
    from chipsec.file import read_file
  File "/app/my_app/chipsec/file.py", line 42, in <module>
    from chipsec.logger import logger
  File "/app/my_app/chipsec/logger.py", line 448, in <module>
    _logger = Logger()
              ^^^^^^^^
  File "/app/my_app/chipsec/logger.py", line 153, in __init__
    logging.config.fileConfig(configPath, defaults={'filename': repr(logPath)[1:-1]})
  File "/usr/local/lib/python3.11/logging/config.py", line 79, in fileConfig
    handlers = _install_handlers(cp, formatters)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/logging/config.py", line 145, in _install_handlers
    h = klass(*args, **kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/logging/__init__.py", line 1181, in __init__
    StreamHandler.__init__(self, self._open())
                                 ^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/logging/__init__.py", line 1213, in _open
    return open_func(self.baseFilename, self.mode,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/app/my_app/logs/FriDec0922-101151'
```

Sae86, 2022-12-09:
@platomav Thanks for reporting, I am currently working on fixing that. Which line did you run when you got that error?

platomav, 2022-12-09:
Any chipsec module import will cause this crash. It can also be seen at the failed unit tests of the MR:

https://github.com/chipsec/chipsec/actions/runs/3655137185/jobs/6176187772

Sae86, 2022-12-09:
PR updated!

platomav, 2022-12-12:
Thanks for the update @Sae86! I can verify that the latest PR #1616 commit 0eee1b41cb123f315ebee386056dc14072bdc86d fixes the logger circular import crash. Feel free to close this one once the PR is merged.

Also, since it's directly related to the chipsec logger and has become a blocker for external projects which import chipsec functionality, you could please also address #1618 before the next release?

Sae86, 2022-12-13:
@platomav PR updated, crash should no longer occur. Of course! Currently looking into #1618

Sae86, 2022-12-19:
PR has been merged, closing. Feel free to reopen if needed. Thanks!

closed 2022-12-19 by Sae86