chipsec/chipsec #1618: Chipsec logger breaks the logs of python logging module

platomav, 2022-12-08:
The "logger" in Chipsec appears to misbehave by breaking the python logging module in a way which causes external (same interpreter instance) logs to not render properly. Here is pseudocode for that behavior:

```
import logging

my_logger = logging.getLogger("my_app")

my_logger.warning('This is a %s', 'test_1')

from chipsec.modules.tools.uefi import <something>

my_logger.warning('This is a %s', 'test_2')

> This is a test_1
> This is a %s
```

The bug appears to have been caused at v1.9.0 (https://github.com/chipsec/chipsec/commit/d91485a6d89acf3c9c152baba794d5ec325816e3), specifically since commit 185b644c059f5813b3b7c02fccfb20c4927fe092. The following line seems to be the root cause (and by extension, "chipsecrecordfactory" class):

https://github.com/chipsec/chipsec/blob/7d97b92a27e3f52175a082e602c4ef75481bd6e2/chipsec/logger.py#L140

Issue split from https://github.com/chipsec/chipsec/issues/1572 to be tracked here instead.

Sae86, 2022-12-19:
PR #1616 has been merged, closing. Feel free to reopen if needed. Thanks!

closed 2022-12-19 by Sae86

platomav, 2023-01-03:
Hi @Sae86 @npmitche,

Thank you for the work so far and happy new year!

I tested the proposed fix bundled with the latest Chipsec release (1.10.0) and the logging problem persists. I have created a new issue with more targeted info (caused under Docker) and provided a reproducible test to validate the requested fix faster.

Please check https://github.com/chipsec/chipsec/issues/1629 for more info.