chipsec/chipsec #1621: KeyError on spidesc parser

0xfede7c8, 2022-12-12:
With the attached SPI flash dump (after extracting):

[Supermicro_bios_firmware_unpacked_00a3a39e78b51d6ecd83bb789431e43d9175a5a1_X11DPS8.A03.zip](https://github.com/chipsec/chipsec/files/10210841/Supermicro_bios_firmware_unpacked_00a3a39e78b51d6ecd83bb789431e43d9175a5a1_X11DPS8.A03.zip)

```bash 
python3 chipsec_util.py --no_driver --skip_config spidesc Supermicro_bios_firmware_unpacked_00a3a39e78b51d6ecd83bb789431e43d9175a5a1_X11DPS8.A03
```

I get:

```bash
################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version  : 1.9.1
[CHIPSEC] Arguments: --no_driver --skip_config spidesc Supermicro_bios_firmware_unpacked_00a3a39e78b51d6ecd83bb789431e43d9175a5a1_X11DPS8.A03

[CHIPSEC] OS      : Linux 4.19.0-22-cloud-amd64 #1 SMP Debian 4.19.260-1 (2022-09-29) x86_64
[CHIPSEC] Python  : 3.7.3 (64-bit)
[CHIPSEC] Helper  : LinuxHelper (None)
[CHIPSEC] Platform: Unrecognized Platform
[CHIPSEC]    CPUID: 406F0
[CHIPSEC]      VID: 65535
[CHIPSEC]      DID: 65535
[CHIPSEC]      RID: 255
[CHIPSEC] PCH     : Unrecognized PCH
[CHIPSEC]      VID: 65535
[CHIPSEC]      DID: 65535
[CHIPSEC]      RID: 255
[CHIPSEC] Executing command 'spidesc' with args ['Supermicro_bios_firmware_unpacked_00a3a39e78b51d6ecd83bb789431e43d9175a5a1_X11DPS8.A03']

[CHIPSEC] Parsing SPI Flash Descriptor from file 'Supermicro_bios_firmware_unpacked_00a3a39e78b51d6ecd83bb789431e43d9175a5a1_X11DPS8.A03'

[spi_fd] Valid SPI flash descriptor found at offset 0x00000000

########################################################
# SPI FLASH DESCRIPTOR
########################################################

+ 0x0000 Reserved : 0xb'FFFFFFFFFFFFFFFFFFFFFFFFFFFFFF'
+ 0x0010 Signature: 0x0FF0A55A
Traceback (most recent call last):
  File "chipsec_util.py", line 215, in <module>
    sys.exit(main())
  File "chipsec_util.py", line 210, in main
    return chipsecMain.main()
  File "chipsec_util.py", line 200, in main
    comm.run()
  File "/chipsec/utilcmd/spidesc_cmd.py", line 52, in run
    parse_spi_flash_descriptor(self.cs, fd)
  File "/chipsec/hal/spi_descriptor.py", line 116, in parse_spi_flash_descriptor
    cs.print_register('FLMAP0', flmap0)
  File "/chipsec/chipset.py", line 1204, in print_register
    reg = self.get_register_def(reg_name)
  File "/chipsec/chipsec/chipset.py", line 809, in get_register_def
    reg_def = self.Cfg.REGISTERS[reg_name]
KeyError: 'FLMAP0'
```

0xfede7c8, 2023-01-04:
I think this was also reported/related to: 

https://github.com/chipsec/chipsec/issues/1262#issuecomment-918469435

npmitche, 2024-09-17:
Better late than never?
So far #2300 has part of the fix. The remaining part of the issue is chipsec is trying to read a register. Without loading a configuration, it doesn't know what/where the register is. With #2300, if you pass in what platform the image is for (eg `-p adl`) It should work.

My next plan is to try and add some code to check a configuration is loaded. If not, print a message asking for the user to pass in what platform it is with -p.

PR #2301 (2024-09-20): Update chipsec_util to check if a cfg is loaded when it's needed
Fixes #1621
In conjunction with #2300
closed 2024-09-20 by npmitche