chipsec/chipsec #1626: can't build kext's asm file with Xcode 13.4.1

kambala-decapitator, 2022-12-15:
```
CompileC /Volumes/kambala/Downloads/chipsec-main/build/lib/drivers/osx/build/chipsec.build/Release/chipsec.build/Objects-normal/arm64e/cpu.o /Volumes/kambala/Downloads/chipsec-main/build/lib/drivers/osx/chipsec/cpu.asm normal arm64e assembler-with-cpp com.apple.compilers.llvm.clang.1_0.compiler (in target 'chipsec' from project 'chipsec')
    cd /Volumes/kambala/Downloads/chipsec-main/build/lib/drivers/osx
    export LANG\=en_US.US-ASCII
    /Applications/Xcode13.4.1.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang -x assembler-with-cpp -target arm64e-apple-macos10.7 -fmessage-length\=120 -fdiagnostics-show-note-include-stack -fmacro-backtrace-limit\=0 -fcolor-diagnostics -nostdinc -fmodules -gmodules -fmodules-prune-interval\=86400 -fmodules-prune-after\=345600 -fbuild-session-file\=/var/folders/r1/j8pmd4951x7_xm15z96k43j80000gn/C/org.llvm.clang/ModuleCache.noindex/Session.modulevalidation -fmodules-validate-once-per-build-session -Wnon-modular-include-in-framework-module -Werror\=non-modular-include-in-framework-module -fno-builtin -Wno-trigraphs -Os -fno-common -mkernel -Wno-missing-field-initializers -Wno-missing-prototypes -Werror\=return-type -Wunreachable-code -Werror\=deprecated-objc-isa-usage -Werror\=objc-root-class -Wno-missing-braces -Wparentheses -Wswitch -Wunused-function -Wno-unused-label -Wno-unused-parameter -Wunused-variable -Wunused-value -Wempty-body -Wuninitialized -Wconditional-uninitialized -Wno-unknown-pragmas -Wno-shadow -Wno-four-char-constants -Wno-conversion -Wconstant-conversion -Wint-conversion -Wbool-conversion -Wenum-conversion -Wno-float-conversion -Wno-non-literal-null-conversion -Wno-objc-literal-conversion -Wshorten-64-to-32 -Wno-newline-eof -DKERNEL -DKERNEL_PRIVATE -DDRIVER_PRIVATE -DAPPLE -DNeXT -isysroot /Applications/Xcode13.4.1.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.3.sdk -fstrict-aliasing -Wdeprecated-declarations -g -Wno-sign-conversion -Wno-infinite-recursion -Wno-comma -Wno-block-capture-autoreleasing -Wno-strict-prototypes -Wno-semicolon-before-method-body -iquote /Volumes/kambala/Downloads/chipsec-main/build/lib/drivers/osx/build/chipsec.build/Release/chipsec.build/chipsec-generated-files.hmap -I/Volumes/kambala/Downloads/chipsec-main/build/lib/drivers/osx/build/chipsec.build/Release/chipsec.build/chipsec-own-target-headers.hmap -I/Volumes/kambala/Downloads/chipsec-main/build/lib/drivers/osx/build/chipsec.build/Release/chipsec.build/chipsec-all-target-headers.hmap -iquote /Volumes/kambala/Downloads/chipsec-main/build/lib/drivers/osx/build/chipsec.build/Release/chipsec.build/chipsec-project-headers.hmap -I/Volumes/kambala/Downloads/chipsec-main/build/lib/drivers/osx/build/Release/include -I/Applications/Xcode13.4.1.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.3.sdk/System/Library/Frameworks/Kernel.framework/PrivateHeaders -I/Applications/Xcode13.4.1.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.3.sdk/System/Library/Frameworks/Kernel.framework/Headers -I/Volumes/kambala/Downloads/chipsec-main/build/lib/drivers/osx/build/chipsec.build/Release/chipsec.build/DerivedSources-normal/arm64e -I/Volumes/kambala/Downloads/chipsec-main/build/lib/drivers/osx/build/chipsec.build/Release/chipsec.build/DerivedSources/arm64e -I/Volumes/kambala/Downloads/chipsec-main/build/lib/drivers/osx/build/chipsec.build/Release/chipsec.build/DerivedSources -F/Volumes/kambala/Downloads/chipsec-main/build/lib/drivers/osx/build/Release -MMD -MT dependencies -MF /Volumes/kambala/Downloads/chipsec-main/build/lib/drivers/osx/build/chipsec.build/Release/chipsec.build/Objects-normal/arm64e/cpu.d --serialize-diagnostics /Volumes/kambala/Downloads/chipsec-main/build/lib/drivers/osx/build/chipsec.build/Release/chipsec.build/Objects-normal/arm64e/cpu.dia -c /Volumes/kambala/Downloads/chipsec-main/build/lib/drivers/osx/chipsec/cpu.asm -o /Volumes/kambala/Downloads/chipsec-main/build/lib/drivers/osx/build/chipsec.build/Release/chipsec.build/Objects-normal/arm64e/cpu.o
/Volumes/kambala/Downloads/chipsec-main/build/lib/drivers/osx/chipsec/cpu.asm:65:10: error: unknown token in expression
    push %rax
         ^
/Volumes/kambala/Downloads/chipsec-main/build/lib/drivers/osx/chipsec/cpu.asm:65:10: error: invalid operand
    push %rax
         ^
...
```

and the list of errors goes on for every asm instruction.

I've also tried building it in Xcode IDE: ordinary build (Debug) worked, but Release build (Archive action) failed with the same errors.

kambala-decapitator, 2022-12-15:
just figured out what's wrong:

> -target **arm64e**-apple-macos10.7

my machine is Intel. Adding `EXCLUDED_ARCHS=arm64e` to the `xcodebuild` invocation removed build error.

npmitche, 2023-01-06:
Can you check to see if this PR fixes it?

https://github.com/chipsec/chipsec/pull/1631

kambala-decapitator, 2023-01-07:
@npmitche thanks, the build error is gone