chipsec/chipsec #1629: Chipsec logger breaks the logs of Python logging under Docker

platomav, 2023-01-03:
The Chipsec logger logic appears to misbehave by breaking the python logging module in a way which causes external (same interpreter instance) logs to not render properly.

This issue is directly related to https://github.com/chipsec/chipsec/issues/1618, but with more targeted info and a reproducible test to validate the requested fix.

Testing against the latest Chipsec version (1.10.0 - d1e3ab2800eac69297e65bce909003a6e301d5e9), the logging error persists but now "None" is shown instead of non-expanded string variables.

It appears the problem is caused when Chipsec is run within [Python-based Docker images](https://hub.docker.com/_/python) and has been reproduced at all compatible Linux distros and Python versions, so the source is definitely Chipsec itself.

In order to expedite the detection and fix validation of the underlying bug, I have created a tiny Docker package which builds Chipsec under Docker and executes a quick logging test script to show the behavior before and after importing any Chipsec module:

[chipsec_test.zip](https://github.com/chipsec/chipsec/files/10337500/chipsec_test.zip)

``` bash
plato@test:~/chipsec_test$ python ./run.py

Building Docker Image...

Sending build context to Docker daemon  4.608kB
Step 1/9 : FROM python:3.11-slim-buster AS test_build
 ---> b5fc9ba9c17b
Step 2/9 : WORKDIR ./
 ---> Using cache
 ---> 9062bf392a71
Step 3/9 : RUN set -xue     && apt-get update -yq     && apt-get install -yq --no-install-recommends         git         gcc         build-essential         python3-dev         python3-setuptools
 ---> Using cache
 ---> f5ac71208398
Step 4/9 : RUN git clone https://github.com/chipsec/chipsec.git chipsec     && (cd chipsec && git checkout d1e3ab2800eac69297e65bce909003a6e301d5e9     && python setup.py build_ext -i --skip-driver)
 ---> Using cache
 ---> 256295795571
Step 5/9 : RUN python -m pip install --root-user-action ignore --upgrade pip     && pip install --root-user-action ignore -r ./chipsec/linux_requirements.txt
 ---> Using cache
 ---> c7c74f1a6976
Step 6/9 : FROM test_build AS test_app
 ---> c7c74f1a6976
Step 7/9 : COPY --from=test_build ./chipsec ./
 ---> Using cache
 ---> 5e303b22cdb4
Step 8/9 : COPY test.py ./
 ---> d7f3ad4ef704
Step 9/9 : ENTRYPOINT ["python", "-u", "-q", "-B", "test.py"]
 ---> Running in b2259375ca54
Removing intermediate container b2259375ca54
 ---> f5718af06145
Successfully built f5718af06145
Successfully tagged log_test:latest

Running Docker Image...

Logging before Chipsec...

    Test log #1 : ABC [DEF]

Logging after Chipsec...

None

plato@test:~/chipsec_test$  
```

The expected output is:

```
    Test log #1 : ABC [DEF]

    Test log #2 : GHI [JKL]
```

not

```
    Test log #1 : ABC [DEF]

None
```

After installing Docker, you can execute the test as such:

`python ./run.py`

npmitche, 2023-01-18:
Hi @platomav Could you try the PR https://github.com/chipsec/chipsec/pull/1646 to see if this fixes this issue? Thanks!

platomav, 2023-01-19:
@npmitche @BrentHoltsclaw 

``` python
--- Logging error ---

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/logging/__init__.py", line 1110, in emit
    msg = self.format(record)
          ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/logging/__init__.py", line 953, in format
    return fmt.format(record)
           ^^^^^^^^^^^^^^^^^^
  File "/app/tester/chipsec/logger.py", line 181, in format
    if record.args[0] is not None and record.args[0] in self.colors:
                                                        ^^^^^^^^^^^
AttributeError: 'chipsecStreamFormatter' object has no attribute 'colors'

Call stack:

  [...]

  File "/app/tester/chipsec/hal/spi_uefi.py", line 353, in build_efi_model
    model = build_efi_tree(data, fwtype)
  File "/app/tester/chipsec/hal/spi_uefi.py", line 293, in build_efi_tree
    fwbin = build_efi_file_tree(fv.Image, fwtype)
  File "/app/tester/chipsec/hal/spi_uefi.py", line 265, in build_efi_file_tree
    fwbin.children = build_efi_tree(fwbin.Image[fwbin.HeaderSize:], fwtype)
  File "/app/tester/chipsec/hal/spi_uefi.py", line 293, in build_efi_tree
    fwbin = build_efi_file_tree(fv.Image, fwtype)
  File "/app/tester/chipsec/hal/spi_uefi.py", line 261, in build_efi_file_tree
    fwbin.children = build_efi_modules_tree(fwtype, fwbin.Image, fwbin.Size, fwbin.HeaderSize, polarity)
  File "/app/tester/chipsec/hal/spi_uefi.py", line 223, in build_efi_modules_tree
    sec.children = build_efi_modules_tree(fwtype, d, len(d), 0, polarity)
  File "/app/tester/chipsec/hal/spi_uefi.py", line 228, in build_efi_modules_tree
    children = build_efi_file_tree(sec.Image[sec.HeaderSize:], fwtype)
  File "/app/tester/chipsec/hal/spi_uefi.py", line 261, in build_efi_file_tree
    fwbin.children = build_efi_modules_tree(fwtype, fwbin.Image, fwbin.Size, fwbin.HeaderSize, polarity)
  File "/app/tester/chipsec/hal/spi_uefi.py", line 184, in build_efi_modules_tree
    sec.children = build_efi_modules_tree(fwtype, sec.Image[sec.DataOffset:], Size - sec.DataOffset, 0, polarity)
  File "/app/tester/chipsec/hal/spi_uefi.py", line 233, in build_efi_modules_tree
    sec.children = build_efi_model(sec.Image[sec.HeaderSize:], fwtype)
  File "/app/tester/chipsec/hal/spi_uefi.py", line 353, in build_efi_model
    model = build_efi_tree(data, fwtype)
  File "/app/tester/chipsec/hal/spi_uefi.py", line 293, in build_efi_tree
    fwbin = build_efi_file_tree(fv.Image, fwtype)
  File "/app/tester/chipsec/hal/spi_uefi.py", line 261, in build_efi_file_tree
    fwbin.children = build_efi_modules_tree(fwtype, fwbin.Image, fwbin.Size, fwbin.HeaderSize, polarity)
  File "/app/tester/chipsec/hal/spi_uefi.py", line 238, in build_efi_modules_tree
    sec.children = build_efi_model(data, fwtype)
  File "/app/tester/chipsec/hal/spi_uefi.py", line 353, in build_efi_model
    model = build_efi_tree(data, fwtype)
  File "/app/tester/chipsec/hal/spi_uefi.py", line 287, in build_efi_tree
    fv = NextFwVolume(data)
  File "/app/tester/chipsec/hal/uefi_fv.py", line 343, in NextFwVolume
    if (ValidateFwVolumeHeader(ZeroVector, FsGuid, FvLength, HeaderLength, ExtHeaderOffset, Reserved, size, CalcSum, Checksum)):
  File "/app/tester/chipsec/hal/uefi_fv.py", line 315, in ValidateFwVolumeHeader
    logger().log_warning("Firmware Volume {{{}}} checksum does not match calculated checksum".format(FsGuid))
  File "/app/tester/chipsec/logger.py", line 334, in log_warning
    self.log(text, level.WARNING)
  File "/app/tester/chipsec/logger.py", line 221, in log
    self.chipsecLogger.log(level.value, text, color)

Message: 'Firmware Volume {0c7d8366-0f10-c784-0000-0066837d0c17} checksum does not match calculated checksum'

Arguments: (Ellipsis,)
```

I am also investigating an issue in which all Chipsec messages are duplicated when I use logging on the code which calls a chipsec module. For example: the message "Message: 'Firmware Volume {0c7d8366-0f10-c784-0000-0066837d0c17} checksum does not match calculated checksum'" is shown twice, once bare and another with (my ?) logging config at the beginning. Chipsec is still intercepting something.

npmitche, 2023-01-19:
Is this from using docker as shown in the op?

platomav, 2023-01-19:
Yes, I am trying through docker. This info is probably more relevant for the weird message behavior (None, intercepted, duplicated) etc.

But the other "missing colors attribute" should be a generic bug. That class does not actually have such an attribute, only a local "colors" variable.

npmitche, 2023-01-19:
Does adding `self.chipsecLogger.propagate = False` to chipsec/logger.py help? 
https://github.com/chipsec/chipsec/pull/1646/commits/cdcc027cb447ea2ce7744bb937453a3d4ea66a61

npmitche, 2023-01-19:
I got a docker setup locally and was able to test it:
```
Successfully built 707cf193aa2f
Successfully tagged log_test:latest

Running Docker Image...

Logging before Chipsec...

    Test log #1 : ABC [DEF]

Logging after Chipsec...

    Test log #2 : GHI [JKL]
```

Looks like adding the `self.chipsecLogger.propagate = False` fixed it as before it was printing "None".

platomav, 2023-01-19:
Great! Did you test in on top of the original commit within the test dockerfile (i.e. d1e3ab2800eac69297e65bce909003a6e301d5e9)? That way, we can be sure that only "propagate" was needed and the issue with "None" was not solved alongside the other PRs/commits since then. If you want you can also check a simple "import logging" followed by "logging.info" or similar to verify that it works even when using the base python logging w/o defining a custom one first.

npmitche, 2023-01-19:
I did not do test with the change just on top of d1e3ab2, mainly because the other changes are needed to resolve #1635. That and I'm not very familiar with docker.

npmitche, 2023-01-19:
Should be fixed with https://github.com/chipsec/chipsec/pull/1646. As always, let us know if you run into anything else.

closed 2023-01-19 by npmitche

platomav, 2023-01-20:
@npmitche

As a followup to yesterday's uncertainty as to what fixed what, I did some quick tests today using the old code for my projects and chipsec, before the fixes from PR #1646:

Before PR #1646 and "propagate=True" (i.e. state when ticket was opened):

``` python
2023-01-20:10:19:00,746 INFO None
2023-01-20:10:19:00,746 INFO None
WARNING: Firmware Volume {0c7d8366-0f10-c784-0000-0066837d0c17} checksum does not match calculated checksum
2023-01-20:10:19:02,114 WARNING WARNING: Firmware Volume {0c7d8366-0f10-c784-0000-0066837d0c17} checksum does not match calculated checksum
2023-01-20:10:19:02,123 WARNING WARNING: Firmware Volume {24548b59-8510-75d2-298b-54240803d18b} checksum does not match calculated checksum
WARNING: Firmware Volume {24548b59-8510-75d2-298b-54240803d18b} checksum does not match calculated checksum
2023-01-20:10:19:02,124 WARNING WARNING: Firmware Volume {d03b0000-e577-28eb-8b42-2003c28b5424} checksum does not match calculated checksum
WARNING: Firmware Volume {d03b0000-e577-28eb-8b42-2003c28b5424} checksum does not match calculated checksum
2023-01-20:10:19:02,164 WARNING WARNING: Firmware Volume {0c7d8366-0f10-ba84-0000-0066837d0c17} checksum does not match calculated checksum
WARNING: Firmware Volume {0c7d8366-0f10-ba84-0000-0066837d0c17} checksum does not match calculated checksum
2023-01-20:10:19:02,305 WARNING WARNING: Firmware Volume {fff12b8d-7696-4c8b-a985-2747075b4f50} checksum does not match calculated checksum
WARNING: Firmware Volume {fff12b8d-7696-4c8b-a985-2747075b4f50} checksum does not match calculated checksum
WARNING: Firmware Volume {fff12b8d-7696-4c8b-a985-2747075b4f50} checksum does not match calculated checksum
2023-01-20:10:19:02,306 WARNING WARNING: Firmware Volume {fff12b8d-7696-4c8b-a985-2747075b4f50} checksum does not match calculated checksum
2023-01-20:10:19:02,328 WARNING WARNING: Firmware Volume {0c7d8366-0f10-c784-0000-0066837d0c17} checksum does not match calculated checksum
WARNING: Firmware Volume {0c7d8366-0f10-c784-0000-0066837d0c17} checksum does not match calculated checksum
WARNING: Firmware Volume {24548b59-8510-75d2-298b-54240803d18b} checksum does not match calculated checksum
2023-01-20:10:19:02,338 WARNING WARNING: Firmware Volume {24548b59-8510-75d2-298b-54240803d18b} checksum does not match calculated checksum
WARNING: Firmware Volume {d03b0000-e577-28eb-8b42-2003c28b5424} checksum does not match calculated checksum
2023-01-20:10:19:02,338 WARNING WARNING: Firmware Volume {d03b0000-e577-28eb-8b42-2003c28b5424} checksum does not match calculated checksum
2023-01-20:10:19:02,378 WARNING WARNING: Firmware Volume {0c7d8366-0f10-ba84-0000-0066837d0c17} checksum does not match calculated checksum
WARNING: Firmware Volume {0c7d8366-0f10-ba84-0000-0066837d0c17} checksum does not match calculated checksum
2023-01-20:10:19:03,192 INFO None
2023-01-20:10:19:03,195 INFO None
2023-01-20:10:19:03,208 INFO None
```

Before PR #1646 but with "propagate=False" (i.e. state when ticket was opened but with propagate disabled):

``` python
2023-01-20:10:09:20,296 INFO None
2023-01-20:10:09:20,296 INFO None
WARNING: Firmware Volume {0c7d8366-0f10-c784-0000-0066837d0c17} checksum does not match calculated checksum
WARNING: Firmware Volume {24548b59-8510-75d2-298b-54240803d18b} checksum does not match calculated checksum
WARNING: Firmware Volume {d03b0000-e577-28eb-8b42-2003c28b5424} checksum does not match calculated checksum
WARNING: Firmware Volume {0c7d8366-0f10-ba84-0000-0066837d0c17} checksum does not match calculated checksum
WARNING: Firmware Volume {fff12b8d-7696-4c8b-a985-2747075b4f50} checksum does not match calculated checksum
WARNING: Firmware Volume {fff12b8d-7696-4c8b-a985-2747075b4f50} checksum does not match calculated checksum
WARNING: Firmware Volume {0c7d8366-0f10-c784-0000-0066837d0c17} checksum does not match calculated checksum
WARNING: Firmware Volume {24548b59-8510-75d2-298b-54240803d18b} checksum does not match calculated checksum
WARNING: Firmware Volume {d03b0000-e577-28eb-8b42-2003c28b5424} checksum does not match calculated checksum
WARNING: Firmware Volume {0c7d8366-0f10-ba84-0000-0066837d0c17} checksum does not match calculated checksum
2023-01-20:10:09:22,796 INFO None
2023-01-20:10:09:22,799 INFO None
2023-01-20:10:09:22,811 INFO None
```

As you can see, by only changing "propagate", the message duplication got fixed but not the "None" on all logs surrounding Chipsec in the same python interpreter instance. Which means that there were actually two issues here:

1. Non-Chipsec logs showing as None at the same interpreter instance --> Fixed by PR #1646
2. Chipsec logs duplicated at the other same interpreter instance loggers --> Fixed by "propagate=False"

``` python
2023-01-20:10:13:16,345 INFO This is a log from my_app, and it works - ABC
2023-01-20:10:13:16,345 INFO This is a log from my_app, and it works - DEF
WARNING: Firmware Volume {0c7d8366-0f10-c784-0000-0066837d0c17} checksum does not match calculated checksum
WARNING: Firmware Volume {24548b59-8510-75d2-298b-54240803d18b} checksum does not match calculated checksum
WARNING: Firmware Volume {d03b0000-e577-28eb-8b42-2003c28b5424} checksum does not match calculated checksum
WARNING: Firmware Volume {0c7d8366-0f10-ba84-0000-0066837d0c17} checksum does not match calculated checksum
WARNING: Firmware Volume {fff12b8d-7696-4c8b-a985-2747075b4f50} checksum does not match calculated checksum
[... cut ...]
WARNING: Firmware Volume {24548b59-8510-75d2-298b-54240803d18b} checksum does not match calculated checksum
WARNING: Firmware Volume {d03b0000-e577-28eb-8b42-2003c28b5424} checksum does not match calculated checksum
WARNING: Firmware Volume {0c7d8366-0f10-ba84-0000-0066837d0c17} checksum does not match calculated checksum
2023-01-20:10:13:18,938 INFO This is a log from my_app, and it works - GHI
2023-01-20:10:13:18,938 INFO This is a log from my_app, and it works - JKL
2023-01-20:10:13:18,989 INFO This is a log from my_app, and it works - MNO
```