chipsec/chipsec #1687: AttributeError: 'Chipset' object has no attribute 'reqs_pch'

platomav, 2023-02-17:
Upon attempting to use Chipsec latest release [(v1.10.2)](Chipsec v1.10.2), the following crash occurs:

``` python
WARNING: Ignoring unsupported platform warning and continue execution.
WARNING: Most results cannot be trusted.
WARNING: Unless a platform independent module is being run, do not file issues against this run.

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version  : 1.10.2
[CHIPSEC] Arguments: --skip_config -i -n -m tools.uefi.scan_image -a generate ./output.json ./input.bin

WARNING: Platform dependent functionality is likely to be incorrect
Traceback (most recent call last):
  File "/home/plato/Downloads/chipsec_demo/chipsec_1.10.2/./chipsec_main.py", line 490, in <module>
    sys.exit(main())
  File "/home/plato/Downloads/chipsec_demo/chipsec_1.10.2/./chipsec_main.py", line 485, in main
    return chipsecMain.main()
  File "/home/plato/Downloads/chipsec_demo/chipsec_1.10.2/./chipsec_main.py", line 461, in main
    print_banner_properties(self._cs, defines.os_version())
  File "/home/plato/Downloads/chipsec_demo/chipsec_1.10.2/chipsec/banner.py", line 85, in print_banner_properties
    logger().log(chipsec_banner_properties(cs, os_version))
  File "/home/plato/Downloads/chipsec_demo/chipsec_1.10.2/chipsec/banner.py", line 60, in chipsec_banner_properties
    include_pch_str = cs.reqs_pch or (cs.reqs_pch is None)
AttributeError: 'Chipset' object has no attribute 'reqs_pch'
```

The temporary workaround is to disable the banner via "-nb" parameter.

frinzell, 2023-02-17:
Please try https://github.com/chipsec/chipsec/pull/1688

platomav, 2023-02-17:
Yeap, fixed at #1688. Feel free to close when merged to main.

frinzell, 2023-02-17:
Thanks for confirming!

platomav, 2023-03-02:
Fixed at [1.10.3](https://github.com/chipsec/chipsec/tree/1.10.3)

closed 2023-03-02 by platomav