chipsec/chipsec #1732: Cannot Setup CHIPSEC

Authey, 2023-04-02:
Hi,

I tended to use iofuzz module on a linux kernel launched by QEMU to test, but I cannot setup the chipsec since a CalledProcessError is raised as below.
[image]
I think this is because the linux kernel has no headers in the system, but when I attempted to install the dependency, it said the package cannot be located, as below.
[image]
I wonder what caused that and do you have any idea to solve this?

Best regards,

Tianyi

drAliRahnema, 2024-02-29:
i have this problem in ubuntu and kali and docker and python docker...

npmitche, 2024-09-17:
Without knowing more about your setup, it seems likely that you're missing the headers and missing the headers would produce an error like this. You could try building your own kernel and exporting the header files accordingly: https://www.kernel.org/doc/html/latest/kbuild/headers_install.html 

Closing issue as it's technically a setup issue, not an issue with the CHIPSEC framework.

closed 2024-09-17 by npmitche