chipsec/chipsec #1736: chipsec 1.10.4 fails to build when installed from pypi.org

rgayon, 2023-04-05:
It looks like the problem is that the `chipsec_tools/compression/Include` directory is not included in the [pip release](https://files.pythonhosted.org/packages/9c/6c/234f0283c79970b3391a3561f1cbfb501f0ccfb4ebb343dcbd995193f501/chipsec-1.10.4.tar.gz)  .

To reproduce:
```
$ virtualenv -p python3 /tmp/c
$ source /tmp/c/bin/activate
$ pip install chipsec
Collecting chipsec
  Downloading chipsec-1.10.4.tar.gz (1.0 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.0/1.0 MB 13.3 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: chipsec
  Building wheel for chipsec (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [304 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-x86_64-cpython-310
      copying chipsec_main.py -> build/lib.linux-x86_64-cpython-310
      copying chipsec_util.py -> build/lib.linux-x86_64-cpython-310
      creating build/lib.linux-x86_64-cpython-310/chipsec
      copying chipsec/exceptions.py -> build/lib.linux-x86_64-cpython-310/chipsec
      copying chipsec/testcase.py -> build/lib.linux-x86_64-cpython-310/chipse
<...SNIP...>
      creating build/lib.linux-x86_64-cpython-310/chipsec/cfg/1022
      copying chipsec/cfg/1022/renoir.xml -> build/lib.linux-x86_64-cpython-310/chipsec/cfg/1022
      copying chipsec/cfg/1022/common.xml -> build/lib.linux-x86_64-cpython-310/chipsec/cfg/1022
      copying chipsec/cfg/template.xml -> build/lib.linux-x86_64-cpython-310/chipsec/cfg
      copying chipsec/cfg/chipsec_cfg.xsd -> build/lib.linux-x86_64-cpython-310/chipsec/cfg
      copying chipsec/modules/tools/secureboot/te.cfg -> build/lib.linux-x86_64-cpython-310/chipsec/modules/tools/secureboot
      copying chipsec/modules/tools/smm/smm_config.ini -> build/lib.linux-x86_64-cpython-310/chipsec/modules/tools/smm
      copying chipsec/modules/tools/uefi/blockedlist.json -> build/lib.linux-x86_64-cpython-310/chipsec/modules/tools/uefi
      copying chipsec/modules/tools/vmm/hv/initial_data.json -> build/lib.linux-x86_64-cpython-310/chipsec/modules/tools/vmm/hv
      running build_ext
      building 'EfiCompressor' extension
      creating build/temp.linux-x86_64-cpython-310
      creating build/temp.linux-x86_64-cpython-310/chipsec_tools
      creating build/temp.linux-x86_64-cpython-310/chipsec_tools/compression
      x86_64-linux-gnu-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -Ichipsec_tools/compression/Include -Ichipsec_tools/compression/Include/Common -Ichipsec_tools/compression/Include/X64 -I/tmp/c/include -I/usr/include/python3.10 -c chipsec_tools/compression/Compress.c -o build/temp.linux-x86_64-cpython-310/chipsec_tools/compression/Compress.o
      chipsec_tools/compression/Compress.c:34:10: fatal error: Compress.h: No such file or directory
         34 | #include "Compress.h"
            |          ^~~~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/x86_64-linux-gnu-gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> chipsec

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
```

npmitche, 2023-04-12:
I would like to keep the version in sync, so I'll fix this on the next release. Thanks for bringing this to our attention! https://github.com/chipsec/chipsec/pull/1743

rgayon, 2023-06-09:
This seems fixed

closed 2023-06-09 by rgayon