chipsec/chipsec #1747: Missing support for LZMAF86 compression

platomav, 2023-04-18:
LZMAF86 is a compression format used in UEFI images, officially supported by EDK2 through GUID D42AE6BD-1352-4BFB-909A-CA72A6EAE889:

https://edk2-docs.gitbook.io/edk-ii-build-specification/5_meta-data_file_specifications/52_tools_def_txt [1]

It is basically a conversion layer on top of a regular LZMA compressed stream, which:

> can improve the compression ratio if the input file is IA32 or X64 PE image. Note: If X64 PE image is built based on GCC44, it may not get the better compression. [1]

Chipsec is not currently aware of the difference between LZMA and LZMAF86 compressed sections, which causes file corruption:

[image]

UEFITool already has support for LZMAF86 detection and proper extraction (https://github.com/LongSoft/UEFITool/issues/197):

[image]

To add support for LZMAF86, you can check for the equivalent GUID and then use EDK2's "LzmaCompress" utility, as such:

`LzmaCompress -q -d --f86 LZMAF86_Compressed.bin -o LZMAF86_Decompressed.bin`

The source code for "LzmaCompress" can be found here:

https://github.com/tianocore/edk2/tree/master/BaseTools/Source/C/LzmaCompress

You can find sample UEFI images with LZMAF86 compressed sections from HPE:

- https://support.hpe.com/connect/s/softwaredetails?language=en_US&softwareId=MTX-b1ffc5df57db42a1b558565848
- https://support.hpe.com/connect/s/softwaredetails?language=en_US&softwareId=MTX_76853ab004dc4f4d85a3f229ce

I have attached one sample UEFI image w/ LZMAF86 as well as the compressed and uncompressed section using UEFITool and LzmaCompress, for reference:

[LZMAF86_rar.zip](https://github.com/chipsec/chipsec/files/11260770/LZMAF86_rar.zip)

BrentHoltsclaw, 2023-04-24:
#1758 appears to work with the limited dataset, can you please verify?

platomav, 2023-04-26:
@BrentHoltsclaw 

Thank you for the effort so far! I'm having trouble verifying whether Chipsec can now properly parse the LZMAF86 section. When it parses the attached sample file (i.e. "H08_1.84_10_23_2020.signed.flash.bin") via:

``` bash
python ./chipsec_main.py --skip_config --no_driver -n -m tools.uefi.scan_image -a generate H08_1.84_10_23_2020.signed.flash.json H08_1.84_10_23_2020.signed.flash.bin`

it does not appear to traverse the volume in question. It detects only 85 efi binaries, because that's how many are included outside the LZMAF86 section. If I give it as input the pre-extracted section (i.e. LZMAF86_Decompressed_UEFITool.bin), it detects an additional 473 binaries, which is correct. Meaning, at the sample file, Chipsec should be detecting a total of 558 efi binaries, not 85.

The referenced samples are included in the attached archive of the ticket.

As for the limited dataset, I can now verify that LZMAF86 is used by HPE in at least 800 different BIOS/UEFI update packages.

BrentHoltsclaw, 2023-04-26:
Interesting I generated the following:
```
[*] Running module: chipsec.modules.tools.uefi.scan_image
[*] Module arguments (3):
['generate', 'H08_1.84_10_23_2020.signed.flash.json', 'H08_1.84_10_23_2020.signed.flash.bin']
[x][ =======================================================================
[x][ Module: Simple list generation/checking for (U)EFI firmware
[x][ =======================================================================
[*] Reading firmware from 'H08_1.84_10_23_2020.signed.flash.bin'...
[*] Generating a list of EFI executables from firmware image...
[*] Found 559 EFI executables in UEFI firmware image 'H08_1.84_10_23_2020.signed.flash.bin'
[*] Creating JSON file 'C:\tmp\tmp\testing\H08_1.84_10_23_2020.signed.flash.json'...
```
[H08_1.84_10_23_2020.signed.flash.zip](https://github.com/chipsec/chipsec/files/11334803/H08_1.84_10_23_2020.signed.flash.zip)

Did you run setup.py prior to your run?

platomav, 2023-04-26:
Weird, I did run setup.py first, but I'm seeing the following:

[image]

[image]

BrentHoltsclaw, 2023-04-26:
@platomav thanks for the pointer.
Missed in setup.py linux and windows both needed to be updated.  Was able to replicate and fix.  When you have time please check again.

platomav, 2023-04-26:
@BrentHoltsclaw 

Excellent, It now works:

[image]

I can also verify that the EFI binaries are correctly detected/hashed now. For example:

- StaticSkuDataDxeNeonCityEPRP (45055A79-B385-4705-A3AC-11CE99A1CB47)
    - UEFITool --> 4FA3BEA454AC784671799747EE6F82C73F6A1A5594E39D21CD382A290C5C3D57
    - Chipsec --> 4FA3BEA454AC784671799747EE6F82C73F6A1A5594E39D21CD382A290C5C3D57

platomav, 2023-04-26:
Did a quick check using the 2nd sample (i.e. U30_1.42_06_20_2018.signed) and Chipsec found 521 EFI binaries. I picked one random EFI binary within the LZMAF86 section and compared the result of UEFITool vs Chipsec:

- PciPlatformDxe (E2441B64-7EF4-41FE-B3A3-8CAA7F8D3017)
    - UEFITool --> 6256A2FB679A3D5C504C82582D9F9BAC8342D00932786CA7D9B6BD8141CC5AF6
    - Chipsec --> 6256A2FB679A3D5C504C82582D9F9BAC8342D00932786CA7D9B6BD8141CC5AF6

Looks good on the 2nd LZMAF86 sample image as well.

platomav, 2023-09-01:
Support for LZMAF86 was added in release 1.11.0 (https://github.com/chipsec/chipsec/commit/00ba41226e76bd1492fb9ef7c194e207ab29ea20)

closed 2023-09-01 by platomav