chipsec/chipsec #1756: 'No module named _lzma' error

zeeshansyed27, 2023-04-24:
I've been trying to run chipec_util uefi command in UEFI shell and I'm getting this error . Please help me resolve this issue I'm new to python and chipsec framework. I've attached the screenshot below.
[image]

frinzell, 2023-04-24:
Related:
[edk2-libc : Py3 UEFI port _lzma module missing from implementation](https://bugzilla.tianocore.org/show_bug.cgi?id=4411)

zeeshansyed27, 2023-04-25:
> Related: [edk2-libc : Py3 UEFI port _lzma module missing from implementation](https://bugzilla.tianocore.org/show_bug.cgi?id=4411)

@frinzell Thanks for the link. Also, my system crashes when I run 'python368 chipsec_util.py cpu info' command. Do you have any idea on this?

GelsonFilho, 2023-04-26:
Hello, follow below, for comparison and documentation of the bug.

In my case, I tried running "python chipsec_util.py uefi var-list" and also got similar errors. The same command, if run in an OS environment (in my case I used Linux Mint 21.1), works normally. The problem happens only when the command is executed in UEFI environment. I'm using the latest version to date (1.10.4).

However "python368 chipsec_util.py cpu info" worked normally returning me the cpu information, I believe that the errors are not necessarily related.

npmitche, 2023-04-26:
@zeeshansyed27  Can you open a new issue for the `cpu info` failure and add as much details as you can?

For this issue specifically, can you try this patch: https://github.com/chipsec/chipsec/pull/1767

npmitche, 2023-04-27:
#1767 has been merged in and release is planned for later this week. Closing.

closed 2023-04-27 by npmitche

zeeshansyed27, 2023-04-27:
@npmitche Thanks for your help. I don't get this error now but I'm getting another set of errors