chipsec/chipsec #1783: chipsec_util.py smi send issue in chipsec 1.10.5 & chipsec 1.10.6

Eric6Wang, 2023-05-09:
Hi Sirs
I am BIOS RD
I am developing a project for DELL and I found an error message when running the following command on chipsec 1.10.5 & 1.10.6: "python chipsec_util.py smi send AA 0x41 0 0 0xbcffff8 0 0 0 0".

I found that the return value of the send_sw_smi function in chipsec-main\chipsec\helper\windows\windowshelper.py was changed from ret = struct.unpack(_smi_msg_t_fmt, out_buf) to ret = struct.unpack(_smi_msg_t_fmt, out_buf)[0].

The error message is as below:

_**[*] [DEBUG] Unable to load helper: dal
[*] [DEBUG] Unable to load helper: efi
[*] [DEBUG] Unable to load helper: linux
[*] [DEBUG] Unable to load helper: osx
[*] [DEBUG] Unable to load helper: rwe
[*] [DEBUG] Unable to load helper: simics
[*] [DEBUG] Unable to load helper: win
[*] [DEBUG] [helper] OS: Windows 10 10.0.19041

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version  : 1.10.6-p0066
[CHIPSEC] Arguments: smi send AA 0x41 0 0 0xbcffff8 0 0 0 0

WARNING: *******************************************************************
WARNING: Chipsec should only be used on test systems!
WARNING: It should not be installed/deployed on production end-user systems.
WARNING: See WARNING.txt
WARNING: *******************************************************************

[CHIPSEC] API mode: using CHIPSEC kernel module API

[CHIPSEC] OS      : Windows 10 10.0.19041 AMD64
[CHIPSEC] Python  : 3.8.6 (64-bit)
[CHIPSEC] Helper  : WindowsHelper (\??\C:\Users\dell\Desktop\chipsec_attached\chipsec-main\chipsec\helper\windows\windows_amd64\chipsec_hlpr.sys)
[CHIPSEC] Platform: Mobile 6th Generation Core Processor (Skylake U)
[CHIPSEC]    CPUID: 406E3
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 1904
[CHIPSEC]      RID: 08
[CHIPSEC] PCH     : PCH-U Premium
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 9D48
[CHIPSEC]      RID: 21

[CHIPSEC] Executing command 'smi' with args ['send', 'AA', '0x41', '0', '0', '0xbcffff8', '0', '0', '0', '0']

[CHIPSEC] Sending SW SMI (code: 0x41, data: 0x00)..
          RAX: 0x0000000000000000 (AX will be overridden with values of SW SMI ports B2/B3)
          RBX: 0x000000000BCFFFF8
          RCX: 0x0000000000000000
          RDX: 0x0000000000000000 (DX will be overridden with 0x00B2)
          RSI: 0x0000000000000000
          RDI: 0x0000000000000000
Return values
Traceback (most recent call last):
  File "chipsec_util.py", line 217, in <module>
    sys.exit(main())
  File "chipsec_util.py", line 212, in main
    return chipsecMain.main()
  File "chipsec_util.py", line 202, in main
    comm.run()
  File "C:\Users\dell\Desktop\1.10.66\chipsec-main\chipsec\utilcmd\interrupts_cmd.py", line 158, in run
    self.func()
  File "C:\Users\dell\Desktop\1.10.66\chipsec-main\chipsec\utilcmd\interrupts_cmd.py", line 142, in smi_send
    self.logger.log("          RAX: {:16X}".format(ret[1]))
TypeError: 'int' object is not subscriptable**_

Thanks.

npmitche, 2023-05-09:
Hi Eric, 
Can you try this PR: https://github.com/chipsec/chipsec/pull/1786 and see if it resolves your issue?

Thanks!

Eric6Wang, 2023-05-10:
Hi npmitche
The result is pass

Thanks.

closed 2023-05-10 by npmitche