chipsec/chipsec #1790: Fallback parsing of detected UEFI volumes within padding/non-UEFI sections

platomav, 2023-05-10:
Hello,

There are various slightly odd UEFI capsules which appear to include UEFI volumes within areas which appear as Non-UEFI or padding:

[image]

Currently, Chipsec does not detect any EFI binaries within such (irregular ?) UEFI images when testing with "tools > uefi > scan_image" module, for example.

Would it be possible to add some form of "fallback" parsing in such cases to detect that valid UEFI volumes are included in non-expected sections and process them as normal, similar to what is shown at the picture being done by UEFITool?

Or maybe some older volume definition or similar is missing, perhaps?

[image]

I am attaching 14 sample capsule images, which should help to narrow this down.

[Non_UEFI.zip](https://github.com/chipsec/chipsec/files/11444699/Non_UEFI.zip)

Thanks!

BrentHoltsclaw, 2023-05-16:
@platomav can you help out with validation, I have a limited number of images to run against currently.

```[*] Found 165 EFI executables in UEFI firmware image '6d271ea0a7c4b0846404790fd9a7791bb3781be2_R101acer.Cap'```

platomav, 2023-05-17:
@BrentHoltsclaw Sure! Thanks for the work so far. I checked the PR against the provided images and saw the following:

*_oem --> Success (14) --> Appears to be correct, upon manual checking
*_R101acer.Cap --> Success (165) --> Appears to be correct, upon manual checking
*_SHW4.Cap --> Failure (0) --> Not sure exactly how many to expect, ~200 maybe (_FVH in both Non-UEFI and Padding)
*_FvNvStorage.fv --> Failure/Ignore (0) --> It appears to be NVRAM only, no EFI binaries (false positive)

Additionally, I found a much newer sample which works partially (450 instead of at least 455):

[233baa0298ff7e9d46479868bbeb45948cbf649f_BIOS.zip](https://github.com/chipsec/chipsec/files/11502298/233baa0298ff7e9d46479868bbeb45948cbf649f_BIOS.zip)

I think this is also a "Padding" case. By manually extracting that padding area and feeding it to Chipsec, it got another 5 EFI binaries:

``` json
{
  "de4a46a4dbf04b40694687c5f3d826db265362973f6fc1740d54a46277d9d5f8": {
    "sha1": "59d7be2ead77825ad9f2a24ff8b93521c57fffc9",
    "guid": "8FEEECF1-BCFD-4A78-9231-4801566B3567",
    "name": "AbsoluteComputraceInstaller"
  },
  "399e5355ee4eed2331fefb3e49864da7d0d8c170ade413bea478a14469b540c2": {
    "sha1": "7e9d6cc0f5f28a020dbba47ecf07febbb0adb405",
    "guid": "4EFC51DA-23A6-4790-A292-4985C7F5CDEF",
    "name": "ComputraceEnablerDxe"
  },
  "e8ed47ff9d955bbb4fc349a7ecb5d5b8474cc20a3e32333f477c8bad433aa503": {
    "sha1": "cb06916d485b5dee6bcea0c70560539e8280edef",
    "guid": "4589CBF3-03F9-4998-9D6F-26343C69562A",
    "name": "ComputraceLoader"
  },
  "d77a47563aa18fcbdb92856cde52ea56531ee667b3dc91b09362ad67ca57e15c": {
    "sha1": "f190b200b1260cdbd3d9cbbdb4dcab9d929f4168",
    "guid": "18578E75-D073-4203-90D2-8788A871E44D",
    "name": "LenovoComputraceSmiServices"
  },
  "ef97bc7d42db1f91567f4dce4df19d783d7b7cb247c1c020067d428d50303f8e": {
    "sha1": "b3d4c94a06b1b5fefee629eaedcc94e4e9baa717",
    "guid": "AC60ED9F-523E-4F5B-94CA-3961346A00BA",
    "name": "LenovoVariableInitDxe"
  }
}
```

Also, is there a reason why the check needs to be triggered for larger unknown sections only (currently 0xFFFF or 64KB)? To avoid extra processing time, perhaps?

BrentHoltsclaw, 2023-05-17:
@platomav, thanks that's helpful.  I need to look into feeding padding info back into the parser.  

Made changes based upon feedback and am currently seeing:
*_oem --> 14
*_R101acer.Cap --> 165
*_SHW4.Cap --> 116
233 --> 450

platomav, 2023-05-17:
@BrentHoltsclaw I checked one _SHW4.cap and found 118, so I decided to check all of them manually (UEFITool report - count PE32/TE, MZ000000 pattern within Non-UEFI/Padding) and generated the following results:

0ed790e4195dae17187f7700f1f89f9e7b0cba43_oem --> 14 (PE) = 14
2f23b45580eacbade578a3c5568dd9e2bec260fb_oem --> 14 (PE) = 14
3c44f4d5c08d903ee25e387997abb9ac0fb5f10a_oem --> 14 (PE) = 14
06b5447e03cc848bde4d9eea8f07fb48d8669aff_oem --> 14 (PE) = 14
25e4b598db924041b98f9743bf403fb8997465cb_oem --> 14 (PE) = 14
804f2283a39bef45a9addb9dcd3a8761ff02a313_oem --> 14 (PE) = 14

0ed790e4195dae17187f7700f1f89f9e7b0cba43_SHW4 --> 118 (PE) + 21 (PE, Padding) = 139
2f23b45580eacbade578a3c5568dd9e2bec260fb_SHW4 --> 116 (PE) + 21 (PE, Padding) = 137
3c44f4d5c08d903ee25e387997abb9ac0fb5f10a_SHW4 --> 116 (PE) + 21 (PE, Padding) = 137
7a94369304134768d01f6c46248dad37ded0d492_SHW4 --> 116 (PE) + 21 (PE, Padding) = 137
25e4b598db924041b98f9743bf403fb8997465cb_SHW4 --> 116 (PE) + 22 (PE, Padding) = 138
804f2283a39bef45a9addb9dcd3a8761ff02a313_SHW4 --> 118 (PE) + 21 (PE, Padding) = 139

6d271ea0a7c4b0846404790fd9a7791bb3781be2_R101acer --> 30 (PE) + 134 (Non-UEFI)

233baa0298ff7e9d46479868bbeb45948cbf649f_BIOS --> 444 (PE) + 9 (TE) + 5 (PE, Padding) = 458

Maybe I missed by accident a few binaries due to the manual process, but they should be fairly close/accurate.

[reports.zip](https://github.com/chipsec/chipsec/files/11502969/reports.zip)

platomav, 2023-05-18:
@BrentHoltsclaw  

Ok, I tested with the latest commit and saw improvements for the "_SHW4" samples (116/118 --> 144/145). The good thing is that this is more than what I manually counted yesterday for these samples (137-139) so maybe I missed something or UEFITool is buggy there as well.

Now, as you verified as well, the "BIOS.cap" one still detects 450 only. I did a quick/dirty test and replaced the EFI parsing line below "EFI_FV_FILETYPE_RAW" check as such:

[image]

This seems to help as now 455 binaries are detected (i.e. the Computrace ones). But I'm not sure at all if this test line is properly implemented. Maybe for the "fallback" logic it is better to call "build_efi_modules_tree" instead of "build_efi_tree" only? At least, from what I can tell, we can assume that these Computrace binaries are found under areas which Chipsec determines as "RAW". Still, my manual check yesterday showed 3 more binaries (458 vs 455) so it could be that something else is missing.

Here are the new results based on last commit and 1 result when checking "BIOS.cap" using the test code mentioned above:

``` json
{
    "06b5447e03cc848bde4d9eea8f07fb48d8669aff_oem.cap.json": 14,
    "0ed790e4195dae17187f7700f1f89f9e7b0cba43_SHW4.Cap.json": 145,
    "0ed790e4195dae17187f7700f1f89f9e7b0cba43_oem.cap.json": 14,
    "233baa0298ff7e9d46479868bbeb45948cbf649f_BIOS.cap.json": 450,
    "233baa0298ff7e9d46479868bbeb45948cbf649f_BIOS.cap_Padding_Non-empty_Padding.pad.json": 5,
    "233baa0298ff7e9d46479868bbeb45948cbf649f_BIOS.cap_RAW_TEST.json": 455,
    "25e4b598db924041b98f9743bf403fb8997465cb_SHW4.Cap.json": 145,
    "25e4b598db924041b98f9743bf403fb8997465cb_oem.cap.json": 14,
    "2f23b45580eacbade578a3c5568dd9e2bec260fb_SHW4.Cap.json": 145,
    "2f23b45580eacbade578a3c5568dd9e2bec260fb_oem.cap.json": 14,
    "3c44f4d5c08d903ee25e387997abb9ac0fb5f10a_SHW4.Cap.json": 144,
    "3c44f4d5c08d903ee25e387997abb9ac0fb5f10a_oem.cap.json": 14,
    "6d271ea0a7c4b0846404790fd9a7791bb3781be2_R101acer.Cap.json": 165,
    "7a94369304134768d01f6c46248dad37ded0d492_SHW4.Cap.json": 144,
    "804f2283a39bef45a9addb9dcd3a8761ff02a313_SHW4.Cap.json": 145,
    "804f2283a39bef45a9addb9dcd3a8761ff02a313_oem.cap.json": 14
}
```

BrentHoltsclaw, 2023-05-18:
@platomav Thanks. I can verify that works.  

Is there a chance that you were seeing duplicate entries within the file:
```
[*] Reading firmware from '233baa0298ff7e9d46479868bbeb45948cbf649f_BIOS.cap'...
[*] Generating a list of EFI executables from firmware image...
[*] Found 455 EFI executables in UEFI firmware image '233baa0298ff7e9d46479868bbeb45948cbf649f_BIOS.cap'
[*] Found 3 duplicate executables
[*] [VERBOSE]   7064efdb151ad9500f1ab4b253d9df645efd5c9b004b30fc5f210da81b48519c
49703981ac5051402ca0a71580fc67a84f9abd656bf8725183886fd792f5040b
64ee09f850e6229a6f54d505de4b3417b7dbe1244f695010289ad15848e3da42
```

platomav, 2023-05-18:
@BrentHoltsclaw

Ah, interesting. Yes, there does seem to exist 3 duplicate binaries within that image, so 455 should be the correct amount to detect. I suppose this last change has now made it possible to correctly parse all those weirder images.

One question: I get a bit confused with the difference between "build_efi_modules_tree" and "build_efi_tree". I think (?) that "build_efi_modules_tree" is capable of calling "build_efi_tree" during its recursions, when needed. So, is there a reason why the code uses "build_efi_tree" instead of "build_efi_modules_tree" in the other padding/raw/non-uefi attempts? If I understand properly, the last-last check (when "fwbin" is None), cannot use "build_efi_modules_tree" because a "fwbin"-like instance is a requirement, but I'm not sure about the previous checks. Maybe the logic will be more flexible with "build_efi_modules_tree"? Is there a cost or probable issue that may occur with such a switch, in your opinion?

BrentHoltsclaw, 2023-05-18:
From my understanding the code will attempt to find a valid FV (build_efi_tree) by looking for the '_FVH' identifier, find a valid FFS_FILE_HEADER (build_efi_file_tree) by overlaying the header within the bytes to see if it looks valid, find a valid Section Header (build_efi_modules_tree) by overlaying the header within the bytes to see if it looks valid.  I'm guessing that using chipsec for with the scan_image module, passing the data to build_efi_modules_tree should be what is needed.  That being said, within the 233* binary, there is a VSS2 and EVSA section that chipsec is not detecting and parsing properly.  I am tempted to say that we should try both functions to see if anything is found and ignore if neither produces any EFI Modules.  Not sure the UEFI Platform Initialization Specification has any additional suggestions? 

Using the above with the 233* binary, the reason the 5 files were missing before is they were within the RAW section which did not have a _FVH identifier when it was passed to build_efi_tree.

When looking at the Non-UEFI data within the 62d* binary, I noticed that UEFITool was identifing the _FVH and that's why I choose the build_efi_tree.  Want this to be changed or add a function that tries both into  put in?

platomav, 2023-05-19:
Yes, you are right. Now that I checked more carefully, the hierarchy is as you say:

- build_efi_tree (FV)
    - build_efi_file_tree (FILE)
        - build_efi_modules_tree (SECTION)

Thing is, 233* (i.e. BIOS.cap) does seem to have an _FVH before the Computrace stuff in RAW:

[image]

So I guess "build_efi_tree" should have worked? Maybe I confused something. I did also test changing to "build_efi_file_tree" at RAW and that also produced 455. So, for that specific case, both "build_efi_file_tree" and "build_efi_modules_tree" work but not "build_efi_tree".

The key word/phrase though is "in this case". Because these are "uncharted territories" and we can't really know what weird concoctions vendors might have come up with which messes/breaks the UEFI spec, it might be that the only catch-all solution is to have a small function which "brute-forces" the functions and combines all the unique children/results (if any). The only downside I foresee is added processing time, when padding/raw/non-uefi areas are encountered. A function will also eliminate the small duplication which exists now at "build_efi_file_tree". 

Regarding 6d2* (i.e. R101acer.Cap), UEFITool seems to have a parsing issue which detects only 0x25 instead of 0x28 bytes for the _FVH header (the first 0x3 zero bytes are lost), so I presume that's why it's shown as "Padding" within a "Non-UEFI" area. Chipsec's more simplistic "search for _FVH pattern" logic probably helped here:

[image]

BrentHoltsclaw, 2023-05-23:
> The key word/phrase though is "in this case". Because these are "uncharted territories" and we can't really know what weird concoctions vendors might have come up with which messes/breaks the UEFI [...]

@platomav @npmitche 

What changes would you like to see within #1796 PR vs a new PR?

platomav, 2023-05-24:
I think the main goal of the PR would be to scan in an open/versatile way the contents of sections which were previously ignored. For example: non-UEFI, RAW, Padding/Uncharted data regions. I believe this is already present in the current PR, with the three added use cases/code blocks.

The question now is what to check these data regions against. Since we cannot know what might be there (i.e. Volume, File, Section ?), the only solution would be to attempt parsing and see what comes up.

The only method which appears to have an actual/signature check before attempting to parse data is "build_efi_tree" (Volumes), which searches for "_FVH" at the very least.

The "build_efi_file_tree" (Files) and "build_efi_file_modules_tree" (Sections) appear to attempt parsing simply based on whether offsets make sense or similar, but I'm not entirely sure, and I honestly lack deeper UEFI structural understanding to be able to assess them.

I'm tempted to say that, for every "unexpected" data region (e.g. Non-UEFI, Padding, RAW), there can be a function call which will try "build_efi_tree" for Volumes, "build_efi_file_tree" for Files and "build_efi_file_modules_tree" for Sections. That should cover all possible weird cases, with two caveats:

- Some processing will be duplicated, increased scan time (in those Non-UEFI/Padding/RAW cases)
    - For example, an obnoxious example would be a RAW area which has a Volume and then a random uncharted File or Section
- Due to the duplicated processing, only the unique children should be kept in the total results of a given data area
    - For example, calling both "build_efi_tree" and "build_efi_file_tree" when only a Volume ends up being present

Just in case I'm confused due to my limited UEFI spec knowledge and the interoperability/capabilities of the relevant code in Chipsec (build_efi_tree, build_efi_file_tree, build_efi_file_modules_tree), does the above make sense to you @BrentHoltsclaw and @npmitche ?

BrentHoltsclaw, 2023-05-26:
@platomav that seems like a reasonable solution.  I will not be able to make the changes until next week.

platomav, 2023-05-26:
Sure thing @BrentHoltsclaw, thank you for the nice work so far!

BrentHoltsclaw, 2023-07-11:
After digging into some details and doing some experiments with the available samples I've found the following when calling all three functions as described above:
  - "build_efi_file_tree" - The adjustments were made within this function so recalling the function produces no new results.
  - When "build_efi_tree" has results
    - It appears these results should be accepted rather than trying to figure out how to merge two trees at different levels because  "build_efi_file_modules_tree" - may produce results, but they seem to be invalid as the type of section is not within standards and the size overlaps with the build_efi_tree results overlap.
  - When "build_efi_tree" has no results
    -  Adding the results from "build_efi_file_modules_tree" seems to fit within the size and standards

Based upon the above I reworked the function to call the three functions and short circuit if results are found within the call.

Let me know if this is reasonable or if you have results which may counter my observations.

platomav, 2023-07-12:
Excellent, sounds good to me. Basically, whenever a function yields no results, and it makes sense to call another which can fit within the size and standards, it can be attempted to hopefully catch those weird cases.

So, if I understand correctly, in the case of an "EFI_FV_FILETYPE_RAW" which is not "NVAR_NRAM_FS_FILE" (i.e. https://github.com/chipsec/chipsec/issues/1790#issuecomment-1552884189), the code would first try "build_efi_tree" and then "build_efi_modules_tree" (if no results were found ?).

As for where to look, I'm thinking "Non-UEFI", "RAW", "Padding" and trailing leftovers (?, unless that is non-UEFI too). Am I missing something?

As for the changes, I don't think they have been pushed to the PR branch yet but if we answer the small questions above and can verify that the new changes do not regress the results from https://github.com/chipsec/chipsec/issues/1790#issuecomment-1552884189, we're good to proceed.

npmitche, 2023-07-14:
@BrentHoltsclaw  were you going to push another update to #1796?

BrentHoltsclaw, 2023-07-19:
> @BrentHoltsclaw were you going to push another update to #1796?

The PR should have the above changes.  Let me know if something is missing in the code or results.

platomav, 2023-07-19:
I think the confusion stems from [this](https://github.com/chipsec/chipsec/issues/1790#issuecomment-1631297477) sentence:

> Based upon the above I reworked the function to call the three functions and short circuit if results are found within the call.

But somehow I missed the change which added the new "efi_data_search" function:

[image]

Tomorrow I will try to verify that the latest PR commit yields the same results as we had seen [here](https://github.com/chipsec/chipsec/issues/1790#issuecomment-1552884189), and if all good, the PR will be ready for review/merge.

platomav, 2023-07-20:
Ok, I tried to test the PR #1796 with the samples from this issue, but I cannot get past https://github.com/chipsec/chipsec/issues/1888.

@npmitche

Can we fix https://github.com/chipsec/chipsec/issues/1888 first? Or maybe you can give me a hint on what I need to hotfix on top of PR #1796, to be able to start testing it.

npmitche, 2023-07-20:
Can you try with this PR: https://github.com/chipsec/chipsec/pull/1892?

platomav, 2023-07-21:
I remember that I tested with -nb (i.e. --no_banner) as well yesterday, and it would still crash.

platomav, 2023-09-01:
@BrentHoltsclaw

So, today I tested the latest PR. Since the PR is not re-based on the latest release (would be nice), I ended up cloning the latest release (https://github.com/chipsec/chipsec/releases/tag/1.12.1) and then replacing the 3 changed files manually. Upon testing of the provided samples, I saw regressions at the "SHW4.Cap" images:

[image]

By "regressions", I mean based on the last time I performed the same test with the PR on May 18th (https://github.com/chipsec/chipsec/issues/1790#issuecomment-1552884189).

I'm re-attaching the samples I'm testing:

[Non_UEFI.zip](https://github.com/chipsec/chipsec/files/12498410/Non_UEFI.zip)

Here are the results from before vs now:

[Non_UEFI_Results_20230901.zip](https://github.com/chipsec/chipsec/files/12498424/Non_UEFI_Results_20230901.zip)

The command I'm using is the following:

`python ./chipsec_main.py --skip_config -i -nl -n -m tools.uefi.scan_image -a generate output.json input.bin`

platomav, 2023-09-14:
@BrentHoltsclaw @npmitche 

Looking at this "regression" issue, mentioned at https://github.com/chipsec/chipsec/issues/1790#issuecomment-1702763645, I investigated the improved "build_efi_file_tree" function from the suggested PR https://github.com/chipsec/chipsec/pull/1796, and noticed two areas in which we are not using the new "efi_data_search" logic.

As a test, I replaced the two remaining "build_efi_modules_tree" calls within "build_efi_file_tree" with "efi_data_search" calls instead, as such (https://github.com/BrentHoltsclaw/chipsec/pull/1/files):

[image]

That seems to work. The, previously attached, "Non_UEFI" sample files all produce the same good results (https://github.com/chipsec/chipsec/issues/1790#issuecomment-1552884189):

``` json
{
    "06b5447e03cc848bde4d9eea8f07fb48d8669aff_oem.cap.json": 14,
    "0ed790e4195dae17187f7700f1f89f9e7b0cba43_SHW4.Cap.json": 145,
    "0ed790e4195dae17187f7700f1f89f9e7b0cba43_oem.cap.json": 14,
    "233baa0298ff7e9d46479868bbeb45948cbf649f_BIOS.cap.json": 450,
    "233baa0298ff7e9d46479868bbeb45948cbf649f_BIOS.cap_Padding_Non-empty_Padding.pad.json": 5,
    "25e4b598db924041b98f9743bf403fb8997465cb_SHW4.Cap.json": 145,
    "25e4b598db924041b98f9743bf403fb8997465cb_oem.cap.json": 14,
    "2f23b45580eacbade578a3c5568dd9e2bec260fb_SHW4.Cap.json": 145,
    "2f23b45580eacbade578a3c5568dd9e2bec260fb_oem.cap.json": 14,
    "3c44f4d5c08d903ee25e387997abb9ac0fb5f10a_SHW4.Cap.json": 144,
    "3c44f4d5c08d903ee25e387997abb9ac0fb5f10a_oem.cap.json": 14,
    "6d271ea0a7c4b0846404790fd9a7791bb3781be2_R101acer.Cap.json": 165,
    "7a94369304134768d01f6c46248dad37ded0d492_SHW4.Cap.json": 144,
    "804f2283a39bef45a9addb9dcd3a8761ff02a313_SHW4.Cap.json": 145,
    "804f2283a39bef45a9addb9dcd3a8761ff02a313_oem.cap.json": 14
}
```

I also tested the PR on a very small dataset of other UEFI image samples and saw no EFI binary metadata changes or regressions. So, https://github.com/chipsec/chipsec/pull/1796 with the fix from https://github.com/BrentHoltsclaw/chipsec/pull/1 seems to work for all "Non_UEFI" samples and for a few "normal" UEFI image samples.

The only observation I made is that these two error messages appear frequently now:

- ERROR: Volume Size not consistent with Block Maps
- ERROR: Volume is corrupted

They occur during "build_efi_file_tree" calls and seem to originate from "uefi_fv" > "GetFvHeader" function.

Are these error messages ok/expected?

BrentHoltsclaw, 2023-12-07:
apologies for the delay.  I misread the above and thought there was farther investigation needed not just the additional lines of code.  I'll make the additions and hopefully we can get this in for the next release.

The errors mentioned make sense in terms of what you are seeing as the new code is making changing the execution path to look for FVHeaders in places that we previously were not (and in most cases where they shouldn't exist).  Perhaps there is a separate issue and PR needed to clean up the error and logging within this code path.  

thoughts @npmitche

npmitche, 2023-12-09:
Yeah, we can look at cleaning up the logging/messaging. Thanks Brent!

platomav, 2023-12-13:
@BrentHoltsclaw @npmitche 

Hi guys, thanks for pushing this to main. I tried to test it today and would always get only 1 EFI binary. Upon investigation, I determined that commit https://github.com/chipsec/chipsec/commit/f530b70174f338527cf8c461b963bef60a5539fa was the problem. It seems that the fix I proposed at https://github.com/chipsec/chipsec/issues/1790#issuecomment-1719184898 was not correctly committed. I created a PR to fix this:

https://github.com/chipsec/chipsec/pull/2016

platomav, 2023-12-18:
Now that PR #2016 has been merged as 2ad9cfb7959a8aa5b49b7d17b7486289bd85cb51, I re-tested the previously attached "Non_UEFI" sample files and all produce the same good results as https://github.com/chipsec/chipsec/issues/1790#issuecomment-1552884189:

``` json
{
    "06b5447e03cc848bde4d9eea8f07fb48d8669aff_oem.cap.json": 14,
    "0ed790e4195dae17187f7700f1f89f9e7b0cba43_SHW4.Cap.json": 145,
    "0ed790e4195dae17187f7700f1f89f9e7b0cba43_oem.cap.json": 14,
    "233baa0298ff7e9d46479868bbeb45948cbf649f_BIOS.cap.json": 450,
    "233baa0298ff7e9d46479868bbeb45948cbf649f_BIOS.cap_Padding_Non-empty_Padding.pad.json": 5,
    "25e4b598db924041b98f9743bf403fb8997465cb_SHW4.Cap.json": 145,
    "25e4b598db924041b98f9743bf403fb8997465cb_oem.cap.json": 14,
    "2f23b45580eacbade578a3c5568dd9e2bec260fb_SHW4.Cap.json": 145,
    "2f23b45580eacbade578a3c5568dd9e2bec260fb_oem.cap.json": 14,
    "3c44f4d5c08d903ee25e387997abb9ac0fb5f10a_SHW4.Cap.json": 144,
    "3c44f4d5c08d903ee25e387997abb9ac0fb5f10a_oem.cap.json": 14,
    "6d271ea0a7c4b0846404790fd9a7791bb3781be2_R101acer.Cap.json": 165,
    "7a94369304134768d01f6c46248dad37ded0d492_SHW4.Cap.json": 144,
    "804f2283a39bef45a9addb9dcd3a8761ff02a313_SHW4.Cap.json": 145,
    "804f2283a39bef45a9addb9dcd3a8761ff02a313_oem.cap.json": 14
}
```

[Non_UEFI_Final_20231219.zip](https://github.com/chipsec/chipsec/files/13709325/Non_UEFI_Final_20231219.zip)

So it should be ok now, nice work! This issue can be closed as completed and verified now. I will open a new issue to cleanup the messages/logging during the "efi_data_search" flow, as suggested above:

https://github.com/chipsec/chipsec/issues/2028

closed 2023-12-18 by platomav