chipsec/chipsec #1815: Why did version 1.10.6 cancel the DCI register status checking function?

Anping-Wang, 2023-05-29:
Why did the new version 1.10.6 cancel DCI status checking function?  Although I found DCI function on 1.10.2, it is not working correctly on intel icelake D platform. Even though I've turned off DCI and the ECTRL register shows off , the 1.10.2 tool still says it's enabled

Anping-Wang, 2023-05-29:
**This is the log of v1.10.2:**
_[*] Running module: chipsec.modules.common.debugenabled
[x][ =======================================================================
[x][ Module: Debug features test
[x][ =======================================================================

[*] Checking IA32_DEBUG_INTERFACE MSR status
[!] Debug Occurred bit set in IA32_DEBUG_INTERFACE MSR

[*] Checking DCI register status
[-] DCI Debug is enabled

[*] Module Results:
[!] IA32_DEBUG_INTERFACE.DEBUG_OCCURRED bit is set.
[-] FAILED: One or more of the debug checks have failed and a debug feature is enabled_

**This is the log of v1.10.6:**
_[*] Running module: chipsec.modules.common.debugenabled
[x][ =======================================================================
[x][ Module: Debug features test
[x][ =======================================================================

[*] Checking IA32_DEBUG_INTERFACE MSR status
[-] CPU debug enable requested by software.
[!] Debug Occurred bit set in IA32_DEBUG_INTERFACE MSR

[*] Module Results:
[!] IA32_DEBUG_INTERFACE.DEBUG_OCCURRED bit is set.
[!] IA32_DEBUG_INTERFACE.ENABLE bit is set.
[-] FAILED: One or more of the debug checks have failed and a debug feature is enabled_

npmitche, 2024-03-26:
Closing due to lack of response.

closed 2024-03-26 by npmitche