chipsec/chipsec #1841: PR #1712 broke chipsec_util.py decode

fishilico, 2023-06-05:
Hello,

Pull Request https://github.com/chipsec/chipsec/pull/1712 changed function `EFIvar_EVSA` to: https://github.com/chipsec/chipsec/blob/f312edc3d27e735350e64aff0de1932ae6edd5cf/chipsec/hal/uefi_platform.py#L942-L945

This broke `chipsec_util.py decode my_spi_rom.bin` (used to decode a SPI flash dump), with:

```text
AttributeError: 'tuple' object has no attribute 'append'
```

... because `variables[name]` is a tuple, not a list. Before the pull request, it was initialized to `[]`, which was fine.

It seems this change was done to try making `EFIvar_EVSA` return a dict of `EfiVariableType` instead of a dict of lists of `EfiVariableType`. But most functions getting EFI variables (such as `getEFIvariables_NVAR`) actually return `Dict[str, List[EfiVariableType]]` instead of `Dict[str, EfiVariableType]`.

Questions:
- Should `variables[name] = (0, None, None, b'', '', 0)` be reverted back to `variables[name] = []`, to fix the bug?
- Should the type of functions getting EFI variables be changed to `Dict[str, List[EfiVariableType]]` instead of `Dict[str, EfiVariableType]`?

fishilico, 2023-06-05:
Hello,
I confirm https://github.com/chipsec/chipsec/pull/1842 fixes my issue. With this, `./chipsec_util.py --helper linuxnativehelper decode spi_rom.bin` (with a file dump from a Lenovo Thinkpad laptop) created several files which seem to contain expected content:

```text
spi_rom.bin.dir/1_A00000-1FFFFFF_BIOS.bin.dir/nvram_evsa.nvram.bin
spi_rom.bin.dir/1_A00000-1FFFFFF_BIOS.bin.dir/nvram_evsa.nvram.dir/LnvOSLicDesc_E5BBF7BE-2417-499B-97DB-39F4896391BC_NV+BS+RT_0.bin
spi_rom.bin.dir/1_A00000-1FFFFFF_BIOS.bin.dir/nvram_evsa.nvram.dir/LnvActivationVar_E5BBF7BE-2417-499B-97DB-39F4896391BC_NV+BS+RT_0.bin
spi_rom.bin.dir/1_A00000-1FFFFFF_BIOS.bin.dir/nvram_evsa.nvram.dir/Test_E5BBF7BE-2417-499B-97DB-39F4896391BC_NV+BS+RT_0.bin
spi_rom.bin.dir/1_A00000-1FFFFFF_BIOS.bin.dir/nvram_evsa.nvram.dir/pubkey_E5BBF7BE-2417-499B-97DB-39F4896391BC_NV+BS+RT_0.bin
spi_rom.bin.dir/1_A00000-1FFFFFF_BIOS.bin.dir/nvram_evsa.nvram.dir/LnvOA3ID_E5BBF7BE-2417-499B-97DB-39F4896391BC_NV+BS+RT_0.bin
spi_rom.bin.dir/1_A00000-1FFFFFF_BIOS.bin.dir/nvram_evsa.nvram.dir/SplashLogoPackage_E5BBF7BE-2417-499B-97DB-39F4896391BC_NV+BS+RT_0.bin
spi_rom.bin.dir/1_A00000-1FFFFFF_BIOS.bin.dir/nvram_evsa.nvram.lst
```

For information, while reading the PR, I added a comment to it: https://github.com/chipsec/chipsec/pull/1842/files#r1218650869.

frinzell, 2023-06-05:
Thanks!  The updates might address your comments.

closed 2023-06-14 by dscott90