chipsec/chipsec #1865: Pull Request #1809 broke chipset_util decode

fishilico, 2023-06-15:
Hello,

Since commit https://github.com/chipsec/chipsec/commit/5ccd4b817e2c84bb9b265c2c3efa1c5e042afa5a (Pull Request https://github.com/chipsec/chipsec/pull/1809), `./chipsec_util.py -p whl decode spi_rom.bin` fails with:

```text
[CHIPSEC] Executing command 'decode' with args ['spi_rom.bin']

[CHIPSEC] Decoding SPI ROM image from a file 'spi_rom.bin'
[CHIPSEC] Found SPI Flash descriptor at offset 0x0 in the binary 'spi_rom.bin'
WARNING: Could not open log file: /git/chipsec/logs/
Traceback (most recent call last):
  File "./chipsec_util.py", line 210, in <module>
    sys.exit(main())
  File "./chipsec_util.py", line 205, in main
    return chipsecMain.main()
  File "./chipsec_util.py", line 190, in main
    comm.run()
  File "chipsec/utilcmd/decode_cmd.py", line 125, in run
    self.decode_rom()
  File "chipsec/utilcmd/decode_cmd.py", line 117, in decode_rom
    self.logger.set_log_file(_orig_logname)
  File "chipsec/logger.py", line 266, in set_log_file
    self.logfile.setFormatter(self.logFormatter)
AttributeError: 'NoneType' object has no attribute 'setFormatter'
```

The buggy code is https://github.com/chipsec/chipsec/blob/09f421639e3a773e6aa2293d41e31ea486efa5cd/chipsec/logger.py#L251-L266

because in https://github.com/chipsec/chipsec/blob/09f421639e3a773e6aa2293d41e31ea486efa5cd/chipsec/utilcmd/decode_cmd.py#L117

the function `decode_rom` tries to restore the log file name, which was empty (it is initialized to `''`). So `os.path.join(LOG_PATH, name)` joins an empty string with the `LOG_PATH` directory and the resulting `self.LOG_FILE_NAME` is a path to the directory, which cannot be opened as a file.

By the way, `set_log_file` uses `name=None` as optional parameter, but this actually never works because calling `os.path.join(LOG_PATH, name)` with `None`  make the code raise an exception:
```text
TypeError: join() argument must be str, bytes, or os.PathLike object, not 'NoneType'
```

TL;DR: Pull Request https://github.com/chipsec/chipsec/pull/1809 changed the semantics of `Logger.set_log_file` and I think callers such as `decode_rom` in `chipsec/utilcmd/decode_cmd.py` has to be updated with the new semantics. What do you think?

BrentHoltsclaw, 2023-06-15:
There is a similar issue within the `uefi decode` commmand. (uefi_command line 311).

Adding a return after the Exception is hit seems to avoid the error but doesn't seem like the best fix., Also agree that the util commands should be updated.
```
     if self.LOG_FILE_NAME: 
         # Open new log file and keep it opened 
         try: 
             # creates FileHandler for log file 
             self.logfile = logging.FileHandler(filename=self.LOG_FILE_NAME, mode='a') 
         except Exception: 
             print(f'WARNING: Could not open log file: {self.LOG_FILE_NAME}')
             return           
         self.chipsecLogger.addHandler(self.logfile)
``` 

@Sae86 what do you think?

Sae86, 2023-06-15:
Thanks @fishilico and @BrentHoltsclaw, wil take a look at it. I agree util needs updates

Sae86, 2023-06-20:
Hi @fishilico and @BrentHoltsclaw,
Please let me know if PR #1871 fixes the issues.
Thanks!

fishilico, 2023-06-23:
Hello, it fixed the issue (now `./chipsec_util.py -p whl decode spi_rom.bin` works again) but the code is still a little buggy. In https://github.com/chipsec/chipsec/blob/1d309370867db61315035d8f10e264588e772122/chipsec/logger.py#L261-L269

... if `logging.FileHandler` raises an exception, `self.logfile` will stay at `None` and `self.chipsecLogger.addHandler(self.logfile)` or `self.logfile.setFormatter(self.logFormatter)` will crash when trying to use it.

Instead, I suggest introducing a `else` such as:

```python
            # Open new log file and keep it opened
            try:
                # creates FileHandler for log file
                self.logfile = logging.FileHandler(filename=self.LOG_FILE_NAME, mode='a')
            except Exception:
                print(f'WARNING: Could not open log file: {self.LOG_FILE_NAME}')   
            else:            
                self.chipsecLogger.addHandler(self.logfile)
                self.logfile.setFormatter(self.logFormatter)
                self.LOG_TO_FILE = True
                self.chipsecLogger.removeHandler(self.logstream)
```

Another option consists in introducting a `return` after `print(f'WARNING: Could not open log file: {self.LOG_FILE_NAME}')`

@Sae86 What do you think?

(By the way, I don't have access to my usual system now which would enable me to properly test this code and submit a Pull Request, which is why I am not able to create a PR to propose the code I wrote here)

Sae86, 2023-06-28:
Thanks @fishilico! I like the else option. Will open a PR with the change.

PR #1882 (2023-07-05): Add else clause to try block for better flow control
Fixes #1865
closed 2023-07-05 by npmitche