chipsec/chipsec #1888: AttributeError: "NoneType" object has no attribute "get_info"

platomav, 2023-07-06:
Hello,

I encountered the following crash at the latest commit (existed before too but with a slightly different crash point):

[image]

Would it be possible to add a test or similar which attempts to run Chipsec main with the various possible parameter combinations? For example: "-skip_config -i -n". This should help to avoid this kind of parameter combination crash in future releases.

BrentHoltsclaw, 2023-07-07:
I am getting similar issues when using `chipsec_util.py -n --skip-config -nl uefi decode` command.  Doing some debugging it appears the cause of the issue is from #1838.

platomav, 2023-09-01:
Based on https://github.com/chipsec/chipsec/pull/1796#issuecomment-1699928873, this should have been fixed at the latest release.

closed 2023-09-01 by platomav