chipsec/chipsec #1906: Issue with "rtclock -a modify" feature in EFI with Chipsec 1.11.0

lifpie, 2023-08-01:
While conducting tests in EFI, using the EFI Shell downloaded from this link: https://github.com/tianocore/edk2/blob/UDK2018/ShellBinPkg/UefiShell/X64/Shell.efi
I think I've encountered a problem related to the "rtclock -a modify" feature in Chipsec 1.11.0.

Before Chipsec 1.11.0, the "rtclock -a modify" feature worked flawlessly on my laptop. However, after updating to Chipsec 1.11.0 and making changes in Python.efi, I observed that the "rtclock -a modify" command no longer functions as expected.

Steps to Reproduce:
1. Download Chipsec 1.11.0 or higher.
2. Follows the instructions in this link https://chipsec.github.io/installation/USBwithUEFIShell.html#installing-chipsec
3. Attempt to use the "rtclock -a modify" feature with the command: python chipsec_main.py -m common.rtclock -a modify.

Expected Behavior:
1. The "rtclock -a modify" feature should execute successfully and generate the appropriate log files.

Actual Behavior:
1. When running the command "chipsec_main.py -m common.rtclock -a modify," the laptop freezes completely, requiring a forced reboot.

Observations:
1. When using the previous version of Chipsec (Python368.efi - https://github.com/chipsec/chipsec/tree/687ba2352e4460d7bbd9b7b53c5d862840f2de5e/__install__/UEFI) with SHA1: 0e5e4959bafea74d4200b76cc2b183c49cdddd6d, the "rtclock -a modify" feature works as expected, and the corresponding logs are generated successfully.
2. The issue seems to be specific to the latest version of Python.efi from main branch with SHA1: df16ff8b30fe5364dd217b16022300faab6f9a9e, since if I change this file to the previous Python368.efi and rename it to "Python.efi", this feature works as expected, and the corresponding logs are generated successfully.
3. Running the "chipsec_main.py -m common.rtclock -a modify" command leads to the laptop freezing during the test.
4. Using the "-vv" option it is possible to see that the test freezes in a specific point, as may be seem in the picture attached
5. I've attached some pictures and the logs generated both with older and newest version of python.efi

This issue is reproducible on my laptop consistently, and I suspect it might be related to a compatibility problem with the latest version of Python.efi.

Please let me know if you need any further information or assistance to investigate this issue. Thank you!

[rtclock.zip](https://github.com/chipsec/chipsec/files/12233378/rtclock.zip)

lifpie, 2023-08-01:
I also tried running -vv in previous python.efi (0e5e4959bafea74d4200b76cc2b183c49cdddd6d) and got the following log:
[2 - chipsec_main.py -vv -m common.rtclock -a modify.log](https://github.com/chipsec/chipsec/files/12233639/2.-.chipsec_main.py.-vv.-m.common.rtclock.-a.modify.log)

When I try to use the same command with the new python.efi (df16ff8b30fe5364dd217b16022300faab6f9a9e) the screen freezes at this momment, and no log is generated:
[image]

npmitche, 2025-02-14:
Hi @lifpie,

Could you try running using the latest version and add the flag `-nl` (no log) to see if it doesn't hang in that case?
```shell
python chipsec_main.py -nl -m common.rtclock -a modify
```