chipsec/chipsec #1928: Add documentation or script for creating UEFI python build

crass, 2023-09-04:
Based on reading the [build documentation for UEFI Python](https://github.com/tianocore/edk2-libc/blob/master/AppPkg/Applications/Python/Python-3.6.8/Py368ReadMe.txt#L59) the prebuilt UEFI Python in this repo probably needed to have non-default modules configured to be built in. Please provide a script or documentation on how this prebuilt was generated.

npmitche, 2024-05-30:
@Sae86 Can you document the process when you find the time to go through it?

npmitche, 2024-07-02:
FYI, EDK-Libc just created github actions scripts to compile. You can see them here: https://github.com/tianocore/edk2-libc/actions

jpshivakavi, 2024-07-09:
We have recently created github actions to build Python interpreter for UEFI using VS2019 and GCC tool chains.
There is also a pre-built release made available through the edk2-libc repo at [https://github.com/tianocore/edk2-libc/releases/tag/v3.6.8.1](https://github.com/tianocore/edk2-libc/releases/tag/v3.6.8.1).

There are also step by step build instructions available for GCC tool chain in this [document](https://github.com/tianocore/edk2-libc/blob/master/AppPkg/Applications/Python/Python-3.6.8/GCCCompilationBKMs.rst)

closed 2024-07-12 by npmitche