chipsec/chipsec #1957: chipsec_util.py smi send issue in chipsec 1.12.3

Eric6Wang, 2023-10-11:
Hi all owers
We encountered the same issue in Chipsec 1.12.3.
FAIL: rc not 0 for cmd: python chipsec_util.py smi send AA 0x41 0 0 0x69ffff8 0 0 0 0
FAIL: rc not 0 for cmd: python chipsec_util.py smi send AA 0x41 0 0 0x6c00007 0 0 0 0

The previously closed issue was the "chipsec_util.py SMI send issue" in Chipsec 1.10.5 and Chipsec 1.10.6  #1783
https://github.com/chipsec/chipsec/issues/1783

Could you help check this issue?

Thanks.

npmitche, 2023-10-11:
Could you share the full output of the command?

Eric6Wang, 2023-10-13:
Hi npmitche,
It is the same file. (chipsec-main\chipsec\helper\windows\windowshelper.py)
It does not import c_ulong from ctypes.

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version  : 1.12.3-p0071
[CHIPSEC] Arguments: smi send AA 0x41 0 0 0x7d00007 0 0 0 0

WARNING: *******************************************************************
WARNING: Chipsec should only be used on test systems!
WARNING: It should not be installed/deployed on production end-user systems.
WARNING: See WARNING.txt
WARNING: *******************************************************************

[CHIPSEC] OS      : Windows 10 10.0.22621 AMD64
[CHIPSEC] Python  : 3.10.5 (64-bit)
[CHIPSEC] Helper  : WindowsHelper (\??\chipsec-1.12.3-p0071\chipsec\helper\windows\windows_amd64\chipsec_hlpr.sys)
[CHIPSEC] Platform: Mobile 8th Generation Core Processor (Kabylake U-Quad Core)
[CHIPSEC]    CPUID: 806EA
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 5914
[CHIPSEC]      RID: 08
[CHIPSEC] PCH     : PCH-U with iHDCP 2.2 Premium
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 9D4E
[CHIPSEC]      RID: 21

[CHIPSEC] Executing command 'smi' with args ['send', 'AA', '0x41', '0', '0', '0x7d00007', '0', '0', '0', '0']

[CHIPSEC] Sending SW SMI (code: 0x41, data: 0x00)..
          RAX: 0x0000000000000000 (AX will be overridden with values of SW SMI ports B2/B3)
          RBX: 0x0000000007D00007
          RCX: 0x0000000000000000
          RDX: 0x0000000000000000 (DX will be overridden with 0x00B2)
          RSI: 0x0000000000000000
          RDI: 0x0000000000000000
Traceback (most recent call last):
  File "chipsec-1.12.3-p0071\chipsec_util.py", line 228, in <module>
    sys.exit(main())
  File "chipsec-1.12.3-p0071\chipsec_util.py", line 223, in main
    return chipsecMain.main()
  File "chipsec-1.12.3-p0071\chipsec_util.py", line 203, in main
    comm.run()
  File "chipsec-1.12.3-p0071\chipsec\utilcmd\interrupts_cmd.py", line 157, in run
    self.func()
  File "chipsec-1.12.3-p0071\chipsec\utilcmd\interrupts_cmd.py", line 140, in smi_send
    ret = self.interrupts.send_SW_SMI(self.thread_id, self.SMI_code_port_value, self.SMI_data_port_value, self._rax, self._rbx, self._rcx, self._rdx, self._rsi, self._rdi)
  File "chipsec-1.12.3-p0071\chipsec\hal\interrupts.py", line 63, in send_SW_SMI
    return self.cs.helper.send_sw_smi(thread_id, SMI_code_data, _rax, _rbx, _rcx, _rdx, _rsi, _rdi)
  File "chipsec-1.12.3-p0071\chipsec\helper\windows\windowshelper.py", line 797, in send_sw_smi
    out_size = c_ulong(out_length)
NameError: name 'c_ulong' is not defined

Thanks.

npmitche, 2023-10-13:
Yep, #1958  should resolve this issue.

Eric6Wang, 2023-10-13:
Hi npmitche 
Could  you provide test  version  package first?
Because the Dell project issue has a time constraint. We need to test first to ensure that the next version of the tool is working properly.
Thank you very much.😀

npmitche, 2023-10-13:
Hi Eric, we don't really have an easy way to share a test version, but if you:
1. Download the latest release zip and extract it
2. Download the updated file: [windowshelper.py](https://raw.githubusercontent.com/chipsec/chipsec/5fd7eb4a658381712c631f796c3290175a391333/chipsec/helper/windows/windowshelper.py) 
3. Replace the file that's in /chipsec/helper/windows/windowshelper.py with the version you just downloaded
4. Try running the SMI command again

This should enable you to test it out.

Eric6Wang, 2023-10-16:
Hi npmitche
Test pass.

Thank you very much.😀

PR #1958 (2023-10-16): Fix import and possible unbound errors
Fixes #1957
closed 2023-10-16 by dscott90