chipsec/chipsec #2063: Support for AMD Raphael CPUs

jvoisin, 2024-02-12:
In the same vein then #825. Raphael looks similar enough to Renoir that passing `-p renoir` gives correct results.

```console
# head -n 28 /proc/cpuinfo                                                                                                                                                                     
processor	: 0
vendor_id	: AuthenticAMD
cpu family	: 23
model		: 113
model name	: AMD Ryzen 7 3700X 8-Core Processor
stepping	: 0
microcode	: 0x8701030
cpu MHz		: 3799.218
cache size	: 512 KB
physical id	: 0
siblings	: 16
core id		: 0
cpu cores	: 8
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 16
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good nopl nonstop_tsc cpuid extd_apicid aperfmperf rapl pni pclmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 x2apic movbe popcnt aes xsave avx f16c rdrand lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw ibs skinit wdt tce topoext perfctr_core perfctr_nb bpext perfctr_llc mwaitx cpb cat_l3 cdp_l3 hw_pstate ssbd mba ibpb stibp vmmcall fsgsbase bmi1 avx2 smep bmi2 cqm rdt_a rdseed adx smap clflushopt clwb sha_ni xsaveopt xsavec xgetbv1 cqm_llc cqm_occup_llc cqm_mbm_total cqm_mbm_local clzero irperf xsaveerptr rdpru wbnoinvd arat npt lbrv svm_lock nrip_save tsc_scale vmcb_clean flushbyasid decodeassists pausefilter pfthreshold avic v_vmsave_vmload vgif v_spec_ctrl umip rdpid overflow_recov succor smca sev sev_es
bugs		: sysret_ss_attrs spectre_v1 spectre_v2 spec_store_bypass retbleed smt_rsb srso
bogomips	: 7599.91
TLB size	: 3072 4K pages
clflush size	: 64
cache_alignment	: 64
address sizes	: 43 bits physical, 48 bits virtual
power management: ts ttp tm hwpstate cpb eff_freq_ro [13] [14]

# ./chipsec_main.py -vv -m common.bios_smi                                                                                                                                                      

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version  : 1.12.8
[CHIPSEC] Arguments: -vv -m common.bios_smi

[*] [DEBUG] [helper] Linux Helper created
[*] [DEBUG] Module /dev/chipsec loaded successfully
[*] [DEBUG] [helper] Linux Helper started/loaded
[*] [HAL] [cpuid] in: EAX=0x00000001, ECX=0x00000000
[*] [HAL] [cpuid] out: EAX=0x00870F10, EBX=0x02100800, ECX=0x7EF8320B, EDX=0x178BFBFF
[*] [DEBUG] [*] Loading device buses..
ERROR: Unknown Platform: VID = 0xFFFF, DID = 0xFFFF, RID = 0xFF, CPUID = 0x870F10
ERROR: Platform is not supported (Unknown Platform: VID = 0xFFFF, DID = 0xFFFF, RID = 0xFF, CPUID = 0x870F10).
WARNING: Platform dependent functionality is likely to be incorrect

[CHIPSEC] OS      : Linux 6.6.13-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.6.13-1 (2024-01-20) x86_64
[CHIPSEC] Python  : 3.11.8 (64-bit)
[CHIPSEC] Helper  : LinuxHelper (/home/jvoisin/Applications/chipsec/chipsec/helper/linux/chipsec.ko)
[CHIPSEC] Platform: Unrecognized Platform
[CHIPSEC]    CPUID: 870F10
[CHIPSEC]      VID: FFFF
[CHIPSEC]      DID: FFFF
[CHIPSEC]      RID: FF
 
[*] Running from /home/jvoisin/Applications/chipsec
[+] loaded chipsec.modules.common.bios_smi
[*] running loaded modules ..

[*] Running module: chipsec.modules.common.bios_smi
[+] imported: chipsec.modules.common.bios_smi
[*] Module path: /home/jvoisin/Applications/chipsec/chipsec/modules/common/bios_smi.py
[!] Required controls not defined for platform.  Skipping module.
Skipping module chipsec.modules.common.bios_smi since it is not applicable in this environment and/or platform
[... 84 lines cut ...]
[CHIPSEC]      DID: FFFF
[CHIPSEC]      RID: FF

 
[+] loaded chipsec.modules.common.bios_smi
[*] running loaded modules ..

[*] Running module: chipsec.modules.common.bios_smi
[!] Required controls not defined for platform.  Skipping module.
Skipping module chipsec.modules.common.bios_smi since it is not applicable in this environment and/or platform

[CHIPSEC] ***************************  SUMMARY  ***************************
[CHIPSEC] Time elapsed            0.001
[CHIPSEC] Modules total           1
[CHIPSEC] Modules failed to run   0:
[CHIPSEC] Modules passed          0:
[CHIPSEC] Modules information     0:
[CHIPSEC] Modules failed          0:
[CHIPSEC] Modules with warnings   0:
[CHIPSEC] Modules not applicable  1:
NOT APPLICABLE: chipsec.modules.common.bios_smi
[CHIPSEC] *****************************************************************

#
```

kerneis-anssi, 2024-04-18:
@JKingGermany Sorry but I deleted  your comment, it was really far too long and somewhat off-topic (mentioning projects unrelated to chipsec). What you want to do seems a bit unclear, I suggest you have a look at the "Rootkits and Bootkits" book to build a better understanding of how bootkits work. https://nostarch.com/rootkits

Let's keep this issue focused on AMD support for more CPU families please. I have a patch supporting more architectures, but I'm also waiting for more information from AMD before opening a PR. Sorry I cannot provide more details for now.

ctmbl, 2024-07-18:
Hi @kerneis-anssi 
is this PR to support more AMD CPUs still on the way?

I'm new to chipsec but really interested by the project and want to add support for other AMD architectures
where should I start? are they publicly available documentation (didn't find any...)? I only have to add another .xml file in `cfg/<some dir>`?

kerneis-anssi, 2024-07-19:
Hi,
sorry for the lack of update there. I am still hoping to get an upstream contribution at some point, but this should'nt block other effort. What can be done in the meantime:
- add more xml files for other architectures: I should definitely contribute mine, or send them your way so that you can test/review them. Let me know what architectures you have in mind please.
- add some Python modules (and the related XML files to define the relevant registers) to check the state of PCI and PSP config, similar to what https://github.com/IOACtive/Platbox is doing.

ctmbl, 2024-07-23:
Hi @kerneis-anssi 
tks for your answer and sorry for the late response I was busy with something else :sweat_smile: 

> add more xml files for other architectures: I should definitely contribute mine, or send them your way so that you can test/review them. Let me know what architectures you have in mind please.

I'll try to do that! however I might struggle a bit at first I'm really new to this kind of low-level configuration
To be precise I have some AMD CPUs I'd want to check:
- AMD EPYC 9354
- AMD EPYC 9124 
- AMD EPYC 7763

> add some Python modules (and the related XML files to define the relevant registers) to check the state of PCI and PSP config, similar to what https://github.com/IOACtive/Platbox is doing.

tks, I'll check the project and see what I can do also!

ctmbl, 2024-07-24:
I noticed that [fwupd](https://github.com/fwupd/fwupd) seems to support AMD CPUs and check some protection mechanisms (I'm thinking at SPI lock) and that you contributed to it (https://github.com/fwupd/fwupd/issues/4094 for example)

Are you able to confirm that it actually supports the CPUs I mentioned above and that its output can be trusted?

Also, could you precise how chipsec would complement the information given by fwupd? can we use some of the configuration already included in this project?

Once again I'm still discovering this world of firmware and low level security so get me right: my questions are essentially about curiosity and confirming what I understand :smiling_face_with_tear: 
In any case, tks a lot for your help!

JKingGermany, 2024-07-25:
Thanks for your answer 

 
 

Gesendet: Donnerstag, 18. Juli 2024 um 17:29 Uhr

ctmbl, 2024-08-13:
Hi @kerneis-anssi

Have you got some news on your side?

On my side I tried to write a chipsec XML config for my  AMD EPYC 9354 but without success...
I feel like the document I found, namely `Preliminary Processor Programming Reference (PPR) for AMD Family 19h Model 19h B1. 55901 Rev 0.53 - May 4, 2023` is not enough. Either I didn't found the right document (NDA maybe?) or my CPU is too much recent? or even more probable I just didn't manage to read it correctly.

In any case I stopped trying but I'm still interested in what you could add regarding AMD support and will closely look at chipsec updates too (I noticed new issues for AMD support opened recently)