chipsec/chipsec #2223: Python setuptools required for building EfiCompressor is incompatible with EWDK

ArtemBaranov, 2024-07-14:
Python's setuptools library from windows_requirements.txt requires Visual Studio installed and it can't be used with Enterprise WDK. It always searches for C:\Program Files (x86)\Microsoft Visual Studio\Installer\vswhere.exe and waits for its output. Since Visual Studio isn't installed (its files are located in the corresponding EWDK folder), building EfiCompressor always fails.

ArtemBaranov, 2024-07-14:
See https://github.com/pypa/setuptools/blob/main/setuptools/msvc.py line 74, _msvc14_find_vc2017

jun2zhou, 2024-08-01:
@ArtemBaranov which MSVC version you are using, would you please try to use the more recent versions test it?
you can refer the versions we are using in the actions ci environments.
https://github.com/chipsec/chipsec/actions/runs/10170877550/job/28130834523
[image]

https://github.com/chipsec/chipsec/actions/runs/10170877550/job/28130837941
[image]

closed 2024-10-29 by chipsec-bbci