chipsec/chipsec #2230: python3 setup.py build_ext -i not working

BrentHoltsclaw, 2024-07-20:
```
[user@red cs]$ python3 setup.py build_ext -i
running build_ext
/home/user/testing/cs/setup.py:146: SetuptoolsDeprecationWarning: setup.py install is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` directly.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
        ********************************************************************************

!!
  _install.initialize_options(self)
Traceback (most recent call last):
  File "/home/user/testing/cs/setup.py", line 237, in <module>
    setup(
  File "/home/user/.local/lib/python3.9/site-packages/setuptools/__init__.py", line 108, in setup
    return distutils.core.setup(**attrs)
  File "/home/user/.local/lib/python3.9/site-packages/setuptools/_distutils/core.py", line 184, in setup
    return run_commands(dist)
  File "/home/user/.local/lib/python3.9/site-packages/setuptools/_distutils/core.py", line 200, in run_commands
    dist.run_commands()
  File "/home/user/.local/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 970, in run_commands
    self.run_command(cmd)
  File "/home/user/.local/lib/python3.9/site-packages/setuptools/dist.py", line 974, in run_command
    super().run_command(command)
  File "/home/user/.local/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
    cmd_obj.ensure_finalized()
  File "/home/user/.local/lib/python3.9/site-packages/setuptools/_distutils/cmd.py", line 111, in ensure_finalized
    self.finalize_options()
  File "/home/user/testing/cs/setup.py", line 79, in finalize_options
    self.set_undefined_options('install', ('skip_driver', 'skip_driver'))
  File "/home/user/.local/lib/python3.9/site-packages/setuptools/_distutils/cmd.py", line 291, in set_undefined_options
    src_cmd_obj.ensure_finalized()
  File "/home/user/.local/lib/python3.9/site-packages/setuptools/_distutils/cmd.py", line 111, in ensure_finalized
    self.finalize_options()
  File "/home/user/.local/lib/python3.9/site-packages/setuptools/command/install.py", line 57, in finalize_options
    super().finalize_options()
  File "/home/user/.local/lib/python3.9/site-packages/setuptools/_distutils/command/install.py", line 407, in finalize_options
    'dist_fullname': self.distribution.get_fullname(),
  File "/home/user/.local/lib/python3.9/site-packages/setuptools/_core_metadata.py", line 266, in get_fullname
    return _distribution_fullname(self.get_name(), self.get_version())
  File "/home/user/.local/lib/python3.9/site-packages/setuptools/_core_metadata.py", line 284, in _distribution_fullname
    canonicalize_version(version, strip_trailing_zero=False),
TypeError: canonicalize_version() got an unexpected keyword argument 'strip_trailing_zero'
[user@red cs]$ python3
Python 3.9.18 (main, Jan 24 2024, 00:00:00)
[GCC 11.4.1 20231218 (Red Hat 11.4.1-3)] on linux
Type "help", "copyright", "credits" or "license" for more information.
```

npmitche, 2024-07-24:
Trying to reproduce. Unable to using RHEL 8.6, Python 3.9.7, Setuptools 71.1.0. 

What version of RHEL were you using?

BrentHoltsclaw, 2024-07-24:
Red Hat Enterprise Linux release 9.4 (Plow)

npmitche, 2024-07-24:
Still cannot duplicate using RHEL 9.4, Python 3.9.18, Setuptools 71.0.3 

If I can get access to the system, maybe I can debug on that system?

npmitche, 2024-07-25:
https://github.com/pypa/setuptools/issues/4501

Solution: `python3 -m pip install --upgrade packaging`

closed 2024-07-25 by npmitche