chipsec/chipsec #2253: Chipsec fails to detect VariableSmm EFI binaries within UEFI image

platomav, 2024-08-06:
Hello,

After the recent changes of #1790 and #2197 by @BrentHoltsclaw , I run a regression test on some UEFI image samples with known results based on earlier Chipsec versions. In some cases (i.e. R440-021400.bin), I noticed that the latest Chipsec (e.g. [1.13.4](https://github.com/chipsec/chipsec/releases/tag/1.13.4)) misses some EFI binaries when scanned via "scan_image" module:

[image]

``` json
"41c0e1fb9269276ec9d5f88057561a57724429ac82ef71a606d8a6a06160877f": {
    "sha1": "3b3f1db18f2976c2b178fece34b042b6946dbee5",
    "guid": "23A089B3-EED5-4AC5-B2AB-43E3298C2343",
    "name": "VariableSmm",
    "type": "S_PE32"
},
"0bf3fc1f2d034b6cce420773e4ab1950348867636f802e29a84e880eb70b57f8": {
    "sha1": "86767ca96822b9a8e80d02c02d1b7246072e2008",
    "guid": "23A089B3-EED5-4AC5-B2AB-43E3298C2343",
    "name": "VariableSmm",
    "type": "S_PE32"
}
```

The issue appears to stem from chipsec/hal/spi_uefi.py > [efi_data_search](https://github.com/chipsec/chipsec/blob/f19760c63c9bb4398debfd92af812ae64f689781/chipsec/hal/spi_uefi.py#L308) (or at least how it gets called via "[build_efi_file_tree](https://github.com/chipsec/chipsec/blob/f19760c63c9bb4398debfd92af812ae64f689781/chipsec/hal/spi_uefi.py#L255)".

As a quick PoC test, I combined the results from "build_efi_tree", "build_efi_file_tree" and "build_efi_modules_tree" within the logic of "efi_data_search" and it managed to find the 2 extra EFI binaries metadata.

I suspect the actual issue is with how "efi_data_search" gets initialized. The "build_efi_tree" and "build_efi_file_tree" functions need only a data input and the fwtype (None is enough for this). However, "build_efi_modules_tree" works differently. It normally requires a data input and fwtype, but also data size and offset. The last two arguments are effectively hard-coded to "len(data)" for the first and "0x0" for the latter. Looking at the logic of "build_efi_modules_tree", the "offset" parameter is usually initialized with the "HeaderSize" of the Image. The size is usually the "Size" attribute of the Image, but I'm not sure if it should be assumed to always be equal to "len(data)".

Maybe the issue is that the "offset" and "size" arguments of "build_efi_modules_tree" are hard-coded during the call of "efi_data_search"? Additionally, it would appear that how "data" buffer is passed to "efi_data_search" matters because in some cases it may need "fwbin.Image" as an input but others "fwbin.Image[HeaderSize:]" or similar.

How can we handle all these possible cases while also retaining the (clear) benefits of the "efi_data_search" method?

[R440-021400.zip](https://github.com/user-attachments/files/16515037/R440-021400.zip)
[R440-021400.bin_BAD.json](https://github.com/user-attachments/files/16515049/R440-021400.bin_BAD.json)
[R440-021400.bin_GOOD.json](https://github.com/user-attachments/files/16515052/R440-021400.bin_GOOD.json)

platomav, 2024-08-27:
To provide more samples for testing the fix, I have now added 4 of them at the OP, all of which lack the VariableSmm EFI binary:

**73d17a462c4999d1415259e15c2ded55457b06ac.BIN**

```json
[
    {
        "name": "VariableSmm",
        "guid": "23A089B3-EED5-4AC5-B2AB-43E3298C2343",
        "sha1": "064e939d8f30553bc71e151d5bee9fdcc199a04f",
        "type": "S_PE32"
    },
    {
        "name": "VariableSmm",
        "guid": "23A089B3-EED5-4AC5-B2AB-43E3298C2343",
        "sha1": "c61245ff69b4d1dea0accaf2bf54056bc95dc808",
        "type": "S_PE32"
    }
]
```

**78195fd9e70d6538b1873fa7b43d56c02036a66e.efi**

```json
[
    {
        "name": "VariableSmm",
        "guid": "23A089B3-EED5-4AC5-B2AB-43E3298C2343",
        "sha1": "b3c490762268d2d743cfa9ea0602826f9fe82d97",
        "type": "S_PE32"
    },
    {
        "name": "VariableSmm",
        "guid": "23A089B3-EED5-4AC5-B2AB-43E3298C2343",
        "sha1": "1240387225b71bbbda904ef341365414e7aab162",
        "type": "S_PE32"
    }
]
```

**BIOS_JHTC7_LN64_1.0.1.BIN**

```json
[
    {
        "name": "VariableSmm",
        "guid": "8307E5DF-33E5-4D68-96A3-ECA40C812D44",
        "sha1": "0d501c0b572bfe0ec73ed19cf13659978d628037",
        "type": "S_PE32"
    }
]
```

**R440-021400.cap**

```json
[
    {
        "name": "VariableSmm",
        "guid": "23A089B3-EED5-4AC5-B2AB-43E3298C2343",
        "sha1": "86767ca96822b9a8e80d02c02d1b7246072e2008",
        "type": "S_PE32"
    },
    {
        "name": "VariableSmm",
        "guid": "23A089B3-EED5-4AC5-B2AB-43E3298C2343",
        "sha1": "3b3f1db18f2976c2b178fece34b042b6946dbee5",
        "type": "S_PE32"
    }
]
```

npmitche, 2024-09-17:
Hi @platomav, could you provide a PR for this?

platomav, 2024-10-01:
Hi @npmitche @BrentHoltsclaw,

I created a PR which solves the aforementioned issue by addressing two Chipsec HAL bugs:

- https://github.com/chipsec/chipsec/pull/2320

The original issue ended up being that there can be EFI binaries which reference/include a FVH within, which would make Chipsec think that it needed to be parsed as a FV instead of FILE or SECTION, thus miss the EFI binary. However, while investigating this, it became clear that Chipsec does not properly check the size of FVs. Moreover, the method which attempts FV, FILE and SECTION modules in succession was not taking into account the input data offset and size, thus ignoring some EFI SECTION modules. I've provided more fix implementation information at the PR. I'm also attaching the raw EFI SECTION (of VariableSmm) which includes the partial FVH, at ~0x14F300, and showcased those issues in the first place.

Could you please take a look?

Thanks

[fwbin_image.zip](https://github.com/user-attachments/files/17213579/fwbin_image.zip)

platomav, 2024-11-05:
[1.13.7 ](https://github.com/chipsec/chipsec/commit/17f3a504ed7d3a5e6c608f05f225aaff88b9459f)

closed 2024-11-05 by platomav