chipsec/chipsec #2257: Failed to start the driver

modz2014, 2024-08-11:
[image]
sc create chipsec binpath="C:\Users\Modz2014\Downloads\chipsec-1.13.4\drivers\windows\chipsec\x64\Debug\chipsec_hlpr\chipsec_hlpr.sys" type= kernel DisplayName="Chipsec driver"
[SC] CreateService SUCCESS

C:\Windows\System32>sc start chipsec
[SC] StartService FAILED 87:

The parameter is incorrect.

npmitche, 2024-08-16:
Please be sure to disable driver signature checking in Windows. For instructions, please see the Chipsec Documentation: https://chipsec.github.io/installation/InstallWindows.html

modz2014, 2024-09-13:
sorry i have been working i will get back to this asap