chipsec/chipsec #2259: pytest fails

sylv-io, 2024-08-14:
Running `pytest' on this project results in the following error:
```
============================= test session starts ==============================
platform linux -- Python 3.12.4, pytest-8.3.2, pluggy-1.5.0
rootdir: /home/sylv/work/git/chipsec
plugins: xdist-3.5.0, cov-5.0.0, typeguard-4.3.0
collected 214 items

tests/hal/test_cpu.py ..........................................         [ 19%]
tests/hardware/test_elitebook_1040_ubuntu.py s                           [ 20%]
tests/hardware/test_x1_carbon_ubuntu.py s                                [ 20%]
tests/hardware/test_z420_ubuntu.py s                                     [ 21%]
tests/helpers/test_linuxhelper.py ........................               [ 32%]
tests/helpers/test_recordhelper.py .                                     [ 32%]
tests/helpers/test_replayhelper.py ................                      [ 40%]
tests/helpers/test_windowshelper.py ...........................          [ 52%]
tests/modules/test_cpu_info.py ...                                       [ 54%]
tests/modules/test_sgx_check.py .                                        [ 54%]
tests/modules/test_tgl_modules.py ..........................             [ 66%]
tests/software/test_acpi.py .........                                    [ 71%]
tests/software/test_cmos.py .                                            [ 71%]
tests/software/test_cs.py ......                                         [ 74%]
tests/software/test_desc.py .                                            [ 74%]
tests/software/test_main.py ......                                       [ 77%]
tests/software/test_msr.py .                                             [ 78%]
tests/software/test_platform.py .                                        [ 78%]
tests/software/test_reg.py ....                                          [ 80%]
tests/software/test_spi.py ..                                            [ 81%]
tests/software/test_tpm_eventlog.py ..                                   [ 82%]
tests/utilcmd/acpi_cmd/test_acpi_cmd.py FF                               [ 83%]
tests/utilcmd/chipset_cmd/test_chipset_cmd.py .                          [ 83%]
tests/utilcmd/cmos_cmd/test_cmos_cmd.py .....                            [ 85%]
tests/utilcmd/config_cmd/test_config_cmd.py ..                           [ 86%]
tests/utilcmd/cpu_cmd/test_cpu_cmd.py .s                                 [ 87%]
tests/utilcmd/desc_cmd/test_desc_cmd.py ..                               [ 88%]
tests/utilcmd/ec_cmd/test_ec_cmd.py ......                               [ 91%]
tests/utilcmd/mmcfg_base_cmd/test_mmcfg_base_cmd.py .                    [ 92%]
tests/utilcmd/mmcfg_cmd/test_mmcfg_cmd.py ....                           [ 93%]
tests/utilcmd/msr_cmd/test_msr_cmd.py ...                                [ 95%]
tests/utilcmd/tpm_cmd/test_tpm_cmd.py ........                           [ 99%]
tests/utilcmd/txt_cmd/test_txt_cmd.py ..                                 [100%]

=================================== FAILURES ===================================
__________________________ TestAcpiUtilcmd.test_list ___________________________

self = <test_acpi_cmd.TestAcpiUtilcmd testMethod=test_list>

    def test_list(self) -> None:
        init_replay_file = os.path.join(get_main_dir(), "tests", "utilcmd", "adlenumerate.json")
        acpi_list_replay_file = os.path.join(get_main_dir(), "tests", "utilcmd", "acpi_cmd", "acpi_cmd_list_1.json")
>       retval = setup_run_destroy_util(init_replay_file, "acpi", "list", util_replay_file=acpi_list_replay_file)

tests/utilcmd/acpi_cmd/test_acpi_cmd.py:36: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
tests/utilcmd/run_chipsec_util.py:53: in setup_run_destroy_util
    retval = run_chipsec_util(csu, util_replay_file)
tests/utilcmd/run_chipsec_util.py:36: in run_chipsec_util
    comm.set_up()
chipsec/utilcmd/acpi_cmd.py:67: in set_up
    self._acpi = ACPI(self.cs)
chipsec/hal/acpi.py:205: in __init__
    self.get_ACPI_table_list()
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <chipsec.hal.acpi.ACPI object at 0x777b7b1cf6e0>

    def get_ACPI_table_list(self) -> Dict[str, List[int]]:
        try:
            # 1. If didn't work, try using get_ACPI_table if a helper implemented
            #    reading ACPI tables via native API which some OS may provide
            # raise UnimplementedAPIError("asdf")
            logger().log_hal("[acpi] Trying to enumerate ACPI tables using get_ACPI_table...")
>           for table_name in self.cs.helper.enum_ACPI_tables():
E           TypeError: 'NoneType' object is not iterable

chipsec/hal/acpi.py:378: TypeError
__________________________ TestAcpiUtilcmd.test_table __________________________

self = <test_acpi_cmd.TestAcpiUtilcmd testMethod=test_table>

    def test_table(self) -> None:
        init_replay_file = os.path.join(get_main_dir(), "tests", "utilcmd", "adlenumerate.json")
        acpi_table_replay_file = os.path.join(get_main_dir(), "tests", "utilcmd", "acpi_cmd", "acpi_cmd_table_1.json")
>       retval = setup_run_destroy_util(init_replay_file, "acpi", "table XSDT", util_replay_file=acpi_table_replay_file)

tests/utilcmd/acpi_cmd/test_acpi_cmd.py:42: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
tests/utilcmd/run_chipsec_util.py:53: in setup_run_destroy_util
    retval = run_chipsec_util(csu, util_replay_file)
tests/utilcmd/run_chipsec_util.py:36: in run_chipsec_util
    comm.set_up()
chipsec/utilcmd/acpi_cmd.py:67: in set_up
    self._acpi = ACPI(self.cs)
chipsec/hal/acpi.py:205: in __init__
    self.get_ACPI_table_list()
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <chipsec.hal.acpi.ACPI object at 0x777b7be11670>

    def get_ACPI_table_list(self) -> Dict[str, List[int]]:
        try:
            # 1. If didn't work, try using get_ACPI_table if a helper implemented
            #    reading ACPI tables via native API which some OS may provide
            # raise UnimplementedAPIError("asdf")
            logger().log_hal("[acpi] Trying to enumerate ACPI tables using get_ACPI_table...")
>           for table_name in self.cs.helper.enum_ACPI_tables():
E           TypeError: 'NoneType' object is not iterable

chipsec/hal/acpi.py:378: TypeError
=============================== warnings summary ===============================
tests/hardware/test_generic.py:23
  /home/sylv/work/git/chipsec/tests/hardware/test_generic.py:23: DeprecationWarning: distro.linux_distribution() is deprecated. It should only be used as a compatibility shim with Python's platform.linux_distribution(). Please use distro.id(), distro.version() and distro.name() instead.
    mSystem = distro.linux_distribution()

-- Docs: https://docs.pytest.org/en/stable/how-to/capture-warnings.html
=========================== short test summary info ============================
FAILED tests/utilcmd/acpi_cmd/test_acpi_cmd.py::TestAcpiUtilcmd::test_list - ...
FAILED tests/utilcmd/acpi_cmd/test_acpi_cmd.py::TestAcpiUtilcmd::test_table
============= 2 failed, 208 passed, 4 skipped, 1 warning in 20.53s =============
```

I don't understand why it's not part of the github testing workflow. Is there a reason for this?

The breaking commit is ed9c6e2, which has been merged here: #2123

sylv-io, 2024-08-14:
cc @npmitche

npmitche, 2024-08-16:
@sylv-io Please try the changes in PR https://github.com/chipsec/chipsec/pull/2267 and let me know if this fixes it for you.

npmitche, 2024-08-16:
We do run unittest in CI, not sure why it wasn't picking up the issue. I'll look into it.

sylv-io, 2024-08-27:
Now the output is even worse than before:
```
============================= test session starts ==============================
platform linux -- Python 3.12.5, pytest-8.3.2, pluggy-1.5.0
rootdir: /home/sylv/work/git/chipsec
plugins: xdist-3.5.0, cov-5.0.0, typeguard-4.3.0
collected 211 items / 5 errors

==================================== ERRORS ====================================
________ ERROR collecting tests/hardware/test_elitebook_1040_ubuntu.py _________
tests/hardware/test_generic.py:24: in <module>
    mSystem = (os_info['name'], os_info['version_id'], os_info['release_codename'])
E   KeyError: 'version_id'

During handling of the above exception, another exception occurred:
tests/hardware/test_elitebook_1040_ubuntu.py:17: in <module>
    from tests.hardware import test_ubuntu
<frozen importlib._bootstrap>:1360: in _find_and_load
    ???
<frozen importlib._bootstrap>:1331: in _find_and_load_unlocked
    ???
<frozen importlib._bootstrap>:935: in _load_unlocked
    ???
/usr/lib/python3.12/site-packages/_pytest/assertion/rewrite.py:174: in exec_module
    exec(co, module.__dict__)
/home/sylv/work/blindspot/git/chipsec/tests/hardware/test_ubuntu.py:20: in <module>
    ???
<frozen importlib._bootstrap>:1360: in _find_and_load
    ???
<frozen importlib._bootstrap>:1331: in _find_and_load_unlocked
    ???
<frozen importlib._bootstrap>:935: in _load_unlocked
    ???
/usr/lib/python3.12/site-packages/_pytest/assertion/rewrite.py:174: in exec_module
    exec(co, module.__dict__)
tests/hardware/test_generic.py:30: in <module>
    mSystem = platform.dist()
E   AttributeError: module 'platform' has no attribute 'dist'
_______________ ERROR collecting tests/hardware/test_generic.py ________________
tests/hardware/test_generic.py:24: in <module>
    mSystem = (os_info['name'], os_info['version_id'], os_info['release_codename'])
E   KeyError: 'version_id'

During handling of the above exception, another exception occurred:
tests/hardware/test_generic.py:30: in <module>
    mSystem = platform.dist()
E   AttributeError: module 'platform' has no attribute 'dist'
________________ ERROR collecting tests/hardware/test_ubuntu.py ________________
tests/hardware/test_generic.py:24: in <module>
    mSystem = (os_info['name'], os_info['version_id'], os_info['release_codename'])
E   KeyError: 'version_id'

During handling of the above exception, another exception occurred:
/usr/lib/python3.12/importlib/__init__.py:90: in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
<frozen importlib._bootstrap>:1387: in _gcd_import
    ???
<frozen importlib._bootstrap>:1360: in _find_and_load
    ???
<frozen importlib._bootstrap>:1331: in _find_and_load_unlocked
    ???
<frozen importlib._bootstrap>:935: in _load_unlocked
    ???
/usr/lib/python3.12/site-packages/_pytest/assertion/rewrite.py:174: in exec_module
    exec(co, module.__dict__)
/home/sylv/work/blindspot/git/chipsec/tests/hardware/test_ubuntu.py:20: in <module>
    ???
<frozen importlib._bootstrap>:1360: in _find_and_load
    ???
<frozen importlib._bootstrap>:1331: in _find_and_load_unlocked
[... 52 lines cut ...]
    exec(co, module.__dict__)
/home/sylv/work/blindspot/git/chipsec/tests/hardware/test_ubuntu.py:20: in <module>
    ???
<frozen importlib._bootstrap>:1360: in _find_and_load
    ???
<frozen importlib._bootstrap>:1331: in _find_and_load_unlocked
    ???
<frozen importlib._bootstrap>:935: in _load_unlocked
    ???
/usr/lib/python3.12/site-packages/_pytest/assertion/rewrite.py:174: in exec_module
    exec(co, module.__dict__)
tests/hardware/test_generic.py:30: in <module>
    mSystem = platform.dist()
E   AttributeError: module 'platform' has no attribute 'dist'
=========================== short test summary info ============================
ERROR tests/hardware/test_elitebook_1040_ubuntu.py - AttributeError: module '...
ERROR tests/hardware/test_generic.py - AttributeError: module 'platform' has ...
ERROR tests/hardware/test_ubuntu.py - AttributeError: module 'platform' has n...
ERROR tests/hardware/test_x1_carbon_ubuntu.py - AttributeError: module 'platf...
ERROR tests/hardware/test_z420_ubuntu.py - AttributeError: module 'platform' ...
!!!!!!!!!!!!!!!!!!! Interrupted: 5 errors during collection !!!!!!!!!!!!!!!!!!!!
============================== 5 errors in 0.29s ===============================
```

I assume this project is only intended to be tested using `python3 -m unittest`.

npmitche, 2024-08-30:
Historically, yes. But I would like to get it working with pytest as well. Can you share the command line you are using? I added pytest run to the GitHub actions and it seems to be working there how I'm running it. (eg https://github.com/chipsec/chipsec/actions/runs/10611695898/job/29411778980 "Run Python pytest") If you're still seeing issues, I would like to get them fixed. But currently I cannot reproduce the issues you're seeing above.

sylv-io, 2024-09-02:
It is also reproducible on Debian unstable (sid):
```
podman pull debian:sid
podman run --rm -it debian:sid bash
# inside the container
apt update
apt install -y build-essential python3-dev python3 python3-setuptools python-is-python3 python3-pytest python3-distro gcc nasm git
git clone https://github.com/chipsec/chipsec.git
cd chipsec
python -m pytest tests
```

You can jump between tags/commits `1.13.5`, `1.13.4`, `ed9c6e2` and `ed9c6e2~` to see the difference in pytest results.

PR #2298 (2024-09-20): Update distro check in test_generic
platform.linux_distribution was deprecated in 3.5 and removed in 3.7.

Fixes #2259
closed 2024-09-24 by chipsec-bbci