chipsec/chipsec #2310: Chipsec cannot read SBREG BAR using mmio module

Bohosul01, 2024-09-25:
Hi,

First of all, yes I have read [this issue](https://github.com/chipsec/chipsec/issues/527) before posting a new thread. I have a similar problem, I'm using an Ice Lake LP platform (495 series chipset) and I'm unable to read SBREG BAR. Basically, Chipsec is unable to unhide P2SB device and it reads SBREG BAR as 0xFFFFFFFF

The thread I linked to above says (from my understand) that the user was running chipsec using another PCH configuration (as the real PCH wasn't supported by chipsec), but that configuration wasn't compatible with the actual PCH.

However, taking a look to my issue, my platform should be recognized and supported by Chipsec because the 495 series PCH documentation was released some years ago, so I can't think of any posible reasons for Chipsec not accomplishing this task.

npmitche, 2024-09-25:
Hi @Bohosul01, Due to security considerations, depending on the platform and where it is in the boot process, it might not possible to unhide the BAR. Generally the easiest thing to do is figure out if you can find the base address and set it in the configuration as a fixed_address. 

Looking at the config file (chipsec/cfg/8086/pch_495.xml), SBREGBAR fixed_address is set. Can you share a log? If you add the `-vv` (very verbose) flag, it can help with debugging too.
https://github.com/chipsec/chipsec/blob/bcc9e506aad93e6690fff980cd6246ca10771b67/chipsec/cfg/8086/pch_495.xml#L40

Bohosul01, 2024-09-26:
Hi, thank you for answering quickly. I checked and, as you say, SBREGBAR address is set at pch_495.xml to 0xFD000000. I have a few questions.

1) Where does this data come from? 495 PCH datasheet doesn't mention that this is the SBREGBAR address. Is this a default address? In case it's a default address, is it valid exclusively for my PCH or is it a general address (valid on more platforms)?

2) The configuration file is there, then why Chipsec reads 0xFFFFFFFF instead of printing the pch_495.xml fixed address when I ask for SBREGBAR address?

3) Do you know any online resource so I can learn about this P2SB "unhide lock"? Is it posible to unlock P2SB unhiding through any BIOS option? Do you know what BIOS / Boot loader component forces P2SB hiding?

Those are general doubts I have, but you don't need to answer if you don't know further. Anyway, I've attached the log file for `python chipsec_util.py -vv mmio read SBREGBAR 0x0 0x4`.

[log.txt](https://github.com/user-attachments/files/17150723/log.txt)

npmitche, 2024-09-27:
1. I don't remember where this data first came from. It's sometimes statically defined in the firmware. Some Linux code might have it statically defined as well. If you do a search for "sbreg.*fixed" (using regex/grep) in the chipsec repo, you'll see that most, if not all, of them have the same fixed address. This address is not guaranteed though. It is also heavily dependent on your motherboard manufacturer as they ultimately control the address. 
2. If you look at the log, you'll see it tries to read the value from the register, then will state it's using the fixed address. (See snip from log below.) Why is it reading all 0xFFs when reading offset 0? Not sure. It's possible it is in a different location on your system. You might also try running in the UEFI shell. There's a small chance the device might not be hidden at that point.
```
[*] [HAL] [pci] reading B/D/F: 0/31/1, offset: 0x10, value: 0xFFFFFFFF  << Read result
WARNING: [mmio] 'limit_field' field not defined for bar, using limit = 0x0
[*] [HAL] [mmio] Using fixed address for SBREGBAR: 0x00000000FD000000 << Ignoring read result and using fixed address
```
3. a. I'm not sure of any resources off hand
b. That is motherboard/firmware specific. It's possible, but very unlikely on a production system.
c. Not sure, sorry.

npmitche, 2024-10-29:
Closing issue. If you have further issues or questions, feel free to re-open or submit a new question or issue.

closed 2024-10-29 by npmitche