chipsec/chipsec #2391: GLK support incomplete

jaredmauch, 2025-01-14:
Looks like GLK platform is missing some configuration or data.  

Based on the comments in glk.xml
`  <info family="atom" detection_value="0x706a0-0x706af"> `

It seems that this hasn't been tested perhaps?  Traceback and some hardware details below.

```
[*] [DEBUG] [helper] Linux Helper created
[*] [DEBUG] [helper] Linux Helper started/loaded
[*] [HAL] [cpuid] in: EAX=0x00000001, ECX=0x00000000
[*] [HAL] [cpuid] out: EAX=0x000706A1, EBX=0x06400800, ECX=0x4FF8EBBF, EDX=0xBFEBFBFF
[*] [DEBUG] [*] Loading device buses..
[!]            Results from this system may be incorrect.
WARNING: Not loading configurations. Platform will remain unknown.

[CHIPSEC] OS      : Linux 6.12.6-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.12.6-1 (2024-12-21) x86_64
[CHIPSEC] Python  : 3.12.8 (64-bit) - Enabled GIL
[CHIPSEC] Helper  : LinuxHelper 
[CHIPSEC] Platform: Unrecognized Platform
[CHIPSEC]    CPUID: FFFFF
[CHIPSEC]      VID: FFFF
[CHIPSEC]      DID: FFFF
[CHIPSEC]      RID: FF
[CHIPSEC] Executing command 'uefi' with args ['tables']

[CHIPSEC] Searching memory for and dumping EFI tables (this may take a minute)..

[*] [HAL] [cpu] Error using get_TSEG() to get SMRAM!
[*] [HAL] [uefi] Searching memory for EFI table with signature 'IBI SYST'...
ERROR: An error occured during the execution of the command!
ERROR: Please run with the debug option for further details
Traceback (most recent call last):
  File "/home/jared/bios/chipsec/chipsec/command.py", line 37, in run
    self.func()
  File "/home/jared/bios/chipsec/chipsec/utilcmd/uefi_cmd.py", line 350, in tables
    self._uefi.dump_EFI_tables()
  File "/home/jared/bios/chipsec/chipsec/hal/uefi.py", line 644, in dump_EFI_tables
    (found, pa, hdr, table, table_buf) = self.find_EFI_System_Table()
                                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jared/bios/chipsec/chipsec/hal/uefi.py", line 589, in find_EFI_System_Table
    return self.find_EFI_Table(EFI_SYSTEM_TABLE_SIGNATURE)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jared/bios/chipsec/chipsec/hal/uefi.py", line 537, in find_EFI_Table
    (tseg_base, tseg_limit, _) = self.cs.cpu.get_TSEG()
                                 ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jared/bios/chipsec/chipsec/hal/cpu.py", line 196, in get_TSEG
    return (tseg_base, tseg_limit, tseg_size)
            ^^^^^^^^^
UnboundLocalError: cannot access local variable 'tseg_base' where it is not associated with a value
```

I've exposed the P2SB and have dumped some of the related interface(s) here but trimmed it back as most of it you likely won't want.

```
wyse02:~/bios/chipsec/logs> sudo lspci -nnnnnvvvvvv
00:00.0 Host bridge [0600]: Intel Corporation Gemini Lake Host Bridge [8086:31f0] (rev 03)
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap- 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	IOMMU group: 1

00:00.3 System peripheral [0880]: Intel Corporation Celeron/Pentium Silver Processor Gaussian Mixture Model [8086:3190] (rev 03)
	Subsystem: Dell Device [1028:080c]

00:02.0 VGA compatible controller [0300]: Intel Corporation GeminiLake [UHD Graphics 605] [8086:3184] (rev 03) (prog-if 00 [VGA controller])
	Subsystem: Dell Device [1028:080c]

00:0d.0 Memory controller [0580]: Intel Corporation Gemini Lake P2SB [8086:3192] (rev 03)
	Subsystem: Dell Device [1028:080c]
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap- 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	IOMMU group: 11
	Region 0: Memory at 80000000 (64-bit, non-prefetchable) [size=16M]

00:0d.2 Serial bus controller [0c80]: Intel Corporation Device [8086:3196] (rev 03)
	Subsystem: Dell Device [1028:080c]
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap- 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	IOMMU group: 12
	Region 0: Memory at 81000000 (32-bit, non-prefetchable) [size=4K]

```

SMBIOS/DMI data:
```
Handle 0x0018, DMI type 4, 48 bytes
Processor Information
        Socket Designation: SOCKET 0
        Type: Central Processor
        Family: Pentium
        Manufacturer: Intel
        ID: A1 06 07 00 FF FB EB BF
        Signature: Type 0, Family 6, Model 122, Stepping 1
```

So with the endian swap EAX=0x000706A1 <-> 0x00 70 06 A1

npmitche, 2025-01-15:
Can you share what version of Chipsec you ran?

dscott90, 2026-10-13:
Haven't heard from reporter. Closing as a legacy issue.

closed 2026-10-13 by dscott90