chipsec/chipsec #2474: Unknown Platform: VID = 0x8086, DID = 0x9A14, RID = 0x01, CPUID = 0x806C1

tallu, 2025-04-28:
Hi. I get the error below when running chipsec_main

# chipsec_main

################################################################
##                                                            ##
##  CHIPSEC: Platform Hardware Security Assessment Framework  ##
##                                                            ##
################################################################
[CHIPSEC] Version  : 1.13.11
[CHIPSEC] Arguments: 

ERROR: Unknown Platform: VID = 0x8086, DID = 0x9A14, RID = 0x01, CPUID = 0x806C1
ERROR: Platform is not supported (Unknown Platform: VID = 0x8086, DID = 0x9A14, RID = 0x01, CPUID = 0x806C1).
WARNING: Platform dependent functionality is likely to be incorrect

[CHIPSEC] OS      : Linux 6.6.88 #8 SMP PREEMPT_DYNAMIC Sun Apr 27 15:12:06 IDT 2025 x86_64
[CHIPSEC] Python  : 3.11.8 (64-bit) - Enabled GIL
[CHIPSEC] Helper  : LinuxHelper (/usr/lib/python3.11/site-packages/chipsec/helper/linux/chipsec.ko)
[CHIPSEC] Platform: Unrecognized Platform
[CHIPSEC]    CPUID: 806C1
[CHIPSEC]      VID: 8086
[CHIPSEC]      DID: 9A14
[CHIPSEC]      RID: 01
 
[*] loading common modules from "/usr/lib/python3.11/site-packages/chipsec/modules/common" ..
[*] No platform specific modules to load
[*] loading modules from "/usr/lib/python3.11/site-packages/chipsec/modules" ..
Traceback (most recent call last):
  File "/usr/bin/chipsec_main", line 33, in <module>
    sys.exit(load_entry_point('chipsec==1.13.11', 'console_scripts', 'chipsec_main')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/chipsec_main.py", line 465, in main
  File "/usr/lib/python3.11/site-packages/chipsec_main.py", line 448, in main
  File "/usr/lib/python3.11/site-packages/chipsec_main.py", line 364, in run_all_modules
  File "/usr/lib/python3.11/site-packages/chipsec_main.py", line 298, in run_loaded_modules
  File "/usr/lib/python3.11/site-packages/chipsec_main.py", line 390, in properties
  File "/usr/lib/python3.11/site-packages/chipsec/chipset.py", line 211, in is_atom
KeyError: 'atom'

npmitche, 2025-04-28:
Hi!
Can you share:

- The steps you did to install and run Chipsec
- The contents of the folder `/usr/lib/python3.11/site-packages/chipsec/cfg/8086`
- What Linux distro/version you're on

Thanks!

tallu, 2025-04-29:
Hi

 

   
I'm using Chipsec with Buildroot.I've built it using a custom Buildroot recipe.If you want, I can share the github for it.
   
My Python version is 3.13. The contents of /usr/lib/python3.13/site-packages/chipsec/cfg/8086 are: adl.xml bdw.xml cfl.xml common.xml glk.xml icl.xml ivb.xml kbl.xml pch_2xx.xml pch_3xx.xml pch_4xxlp.xml pch_5xxlp.xml pch_7x.xml pch_c61x.xml qrk.xml sfdp.xml snb.xml tpm12.xml apl.xml  bdx.xml cht.xml dnv.xml hsw.xml icx.xml ivt.xml mtl.xml pch_3xxlp.xml pch_495.xml pch_4xx.xml pch_6xxP.xml pch_8x.xml pch_c620.xml rkl.xml skl.xml tglh.xml txt.xml avn.xml byt.xml cml.xml ehl.xml hsx.xml iommu.xml jkt.xml pch_1xx.xml pch_3xxop.xml pch_4xxh.xml pch_5xxh.xml  pch_6xxS.xml pch_c60x.xml pmc_i440fx.xml rpl.xml skx.xml tglu.xml whl.xml

npmitche, 2025-05-02:
Yes, can you please share the buildroot/recipe? 

Based on what's in the 8086 folder, it should be working right. The platform should be identified as TGLU (pulling from the tglu.xml file) and the "atom" key should be populated. Something's not looking in the correct folder is my guess. 

Could you also share how you ran chipsec?  And the last thing that could be helpful would be a very verbose log "python3 chipsec_main.py -vv"

Thanks!

tallu, 2025-05-11:
Hi

Here is the Buildroot project
https://github.com/tallu/buildroot-2024.11-chipsec#

The TGLU platform is currently not available, so I repeated the test on a CFL platform (i5-8365UE CPU) and got a similar error:

ERROR: Unknown Platform: VID = 0x8086, DID = 0x3E34, RID = 0x0C, CPUID = 0x806EC
ERROR: Platform is not supported (Unknown Platform: VID = 0x8086, DID = 0x3E34, RID = 0x0C, CPUID = 0x806EC).

Buildroot installs compiled Python, so I run "chipsec_main -vv". This is what I got (on a CFL platform):

[*] [DEBUG] [helper] Linux Helper created
[*] [DEBUG] Module for /dev/chipsec unloaded successfully
[*] [DEBUG] Module /dev/chipsec loaded successfully
[*] [DEBUG] [helper] Linux Helper started/loaded
[*] [HAL] [cpuid] in: EAX=0x00000001, ECX=0x00000000
[*] [HAL] [cpuid] out: EAX=0x000806EC, EBX=0x00100800, ECX=0x7FFAFBFF, EDX=0xBFEBFBFF
[*] [DEBUG] [*] Loading device buses..
ERROR: Unknown Platform: VID = 0x8086, DID = 0x3E34, RID = 0x0C, CPUID = 0x806EC
ERROR: Platform is not supported (Unknown Platform: VID = 0x8086, DID = 0x3E34, RID = 0x0C, CPUID = 0x806EC).
WARNING: Platform dependent functionality is likely to be incorrect

Regards,
Tal

npmitche, 2025-05-29:
Hi, yes. I did see it, though I have not had time to triage the root issue.

Based on what I know, it looks like maybe buildroot is not pulling all of the needed files from chipsec, or the folder structure is malformed in the final state. 

We are working on building a small Linux distro with chipsec included that you can create a live boot image with. You can find the image with the last release (1.13.12) here: https://github.com/chipsec/chipsec/releases/download/1.13.12/arch-linux-image.zip

We are working on getting one with the latest 1.13.13 release. 

I'm not sure if that'll satisfy your requirements for a Linux image, but if it does, that should resolve the issue.

Cheers,
Nathaniel

tallu, 2025-06-04:
Hi

I will try the ISO.
What files and file structure does chipsec assume?

Regards,
Tal

tallu, 2025-06-04:
Hi

Copying the entire contents of lib.linux-x86_64-cpython-312 folder to /usr/lib/python3.12/site-packages/ solved it.
I'll try to look into chipsec's install command.

tallu, 2025-06-06:
My Buildroot was configured to install pyc only.
When I configured it to install both py and pyc chipsec_main worked.
So, I guess that chipsec_main depends on some py files. Why?

npmitche, 2026-10-12:
I'm not sure. It sounds like some py files are not being compiled. If you're still wanting to resolve this, my suggestion would be to compare the compiled python files with what's in py files in the repo and see what's missing and see if you can force a compilation before generating the image. 

Closing as this isn't specifically a chipsec issue, but a Buildroot integration issue.

closed 2026-10-12 by npmitche