chipsec/chipsec #2543: Bad return status for module build on kernel

Bill-Joe-Tech, 2025-07-25:
After running a system upgrade, I am presented with: 

```
(2/3) Install DKMS modules
==> dkms install --no-depmod chipsec-dkms/git -k 6.15.7-arch1-1
Deprecated feature: CLEAN (/usr/src/chipsec-dkms-git/dkms.conf)

Error! Bad return status for module build on kernel: 6.15.7-arch1-1 (x86_64)
Consult /var/lib/dkms/chipsec-dkms/git/build/make.log for more information.
==> WARNING: `dkms install --no-depmod chipsec-dkms/git -k 6.15.7-arch1-1' exited 10
```

Here is `/var/lib/dkms/chipsec-dkms/git/build/make.log`:

```
DKMS (dkms-3.2.1) make.log for chipsec-dkms/git for kernel 6.15.7-arch1-1 (x86_64)
Fri Jul 25 02:26:21 PM CDT 2025

Building module(s)
# command: make -j8 KERNELRELEASE=6.15.7-arch1-1 -C . KSRC=/lib/modules/6.15.7-arch1-1/build
make: Entering directory '/var/lib/dkms/chipsec-dkms/git/build'
make -C /lib/modules/6.15.7-arch1-1/build M=/var/lib/dkms/chipsec-dkms/git/build clean
make[1]: Entering directory '/usr/lib/modules/6.15.7-arch1-1/build'
make[2]: Entering directory '/var/lib/dkms/chipsec-dkms/git/build'
make[2]: Leaving directory '/var/lib/dkms/chipsec-dkms/git/build'
make[1]: Leaving directory '/usr/lib/modules/6.15.7-arch1-1/build'
make -C /lib/modules/6.15.7-arch1-1/build M=/var/lib/dkms/chipsec-dkms/git/build modules
make[1]: Entering directory '/usr/lib/modules/6.15.7-arch1-1/build'
make[2]: Entering directory '/var/lib/dkms/chipsec-dkms/git/build'
  CC [M]  chipsec_km.o
  NASM_64 amd64/cpu.o
chipsec_km.c:262:1: warning: no previous prototype for ‘ReadPCICfg’ [-Wmissing-prototypes]
  262 | ReadPCICfg(
      | ^~~~~~~~~~
chipsec_km.c:280:1: warning: no previous prototype for ‘WritePCICfg’ [-Wmissing-prototypes]
  280 | WritePCICfg(
      | ^~~~~~~~~~~
chipsec_km.c:297:1: warning: no previous prototype for ‘WriteIOPort’ [-Wmissing-prototypes]
  297 | WriteIOPort(
      | ^~~~~~~~~~~
chipsec_km.c:309:1: warning: no previous prototype for ‘ReadIOPort’ [-Wmissing-prototypes]
  309 | ReadIOPort(
      | ^~~~~~~~~~
chipsec_km.c:721:6: warning: no previous prototype for ‘print_stat’ [-Wmissing-prototypes]
  721 | void print_stat(efi_status_t stat)
      |      ^~~~~~~~~~
  LD [M]  chipsec.o
chipsec.o: warning: objtool: .text+0x13d0: unannotated intra-function call
make[4]: *** [/usr/lib/modules/6.15.7-arch1-1/build/scripts/Makefile.build:408: chipsec.o] Error 255
make[4]: *** Deleting file 'chipsec.o'
make[3]: *** [/usr/lib/modules/6.15.7-arch1-1/build/Makefile:2003: .] Error 2
make[2]: *** [/usr/lib/modules/6.15.7-arch1-1/build/Makefile:248: __sub-make] Error 2
make[2]: Leaving directory '/var/lib/dkms/chipsec-dkms/git/build'
make[1]: *** [Makefile:248: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.15.7-arch1-1/build'
make: *** [Makefile:30: chipsec] Error 2
make: Leaving directory '/var/lib/dkms/chipsec-dkms/git/build'

# exit code: 2
# elapsed time: 00:00:03
----------------------------------------------------------------
```

npmitche, 2025-07-25:
Hi! If you use the latest on main, this issue should be resolved. It should be fixed with PR #2532.

Please let me know if that's not the case.

Bill-Joe-Tech, 2025-07-26:
The issue was on  `chipsec 5:1.13.14.r3.gb918f89-1` after running `yay -Qv chipsec | tail -n 1`

Bill-Joe-Tech, 2025-07-27:
The issue has been fixed with simple reboot and system update, thanks!

closed 2026-09-19 by npmitche